alienatedsec/yi-hack-v5 #2: No recording being made

Shourai, 2021-04-06:
Having disabled cloud and enabled recording without cloud, no recordings are saved in `/tmp/sd/`
The device in question is a `48US`

This is the contents of `system.conf`
```
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
PROXYCHAINSNG=no
CHECK_UPDATES=yes
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=no
NTPD=yes
NTP_SERVER=pool.ntp.org
ONVIF=no
ONVIF_WSDD=no
RTSP=yes
RTSP_STREAM=high
```

Furthermore I can confirm SSH, FTP and HTTP to be working.

U-Sus, 2021-04-06:
1st) enable cloud
2st) BACKUP ALL YOUR SD
3st) format sd using yi app
4st) cam should NOT boot. turn off, restore sd backup, put sd in cam then turn on cam.
5st) disable cloud

Shourai, 2021-04-07:
Thanks @U-Sus, 
after formatting the sdcard with the yi app it did leave some recording in the record folder.
But after disabling cloud and enabling record without cloud it doesn't record anything.
I will fiddle around a little more.
`mp4record` is running though.

Furthermore I was wondering, when the SD card gets full does it automatically remove the earliest recordings?

alienatedsec, 2021-04-07:
@Shourai next release is aiming to remove early recordings

Shourai, 2021-04-08:
@alienatedsec & @U-Sus,
the issue seems to be  `watch_process` not running when you disable cloud.
https://github.com/alienatedsec/yi-hack-v5/blob/ddb61d6b192cb3ffac2a416bf1374b8ec2e5c3d9/src/static/static/home/yi-hack-v5/script/system.sh#L108

If I run this binary it seems to record.
Though I am not sure what this binary does.

alienatedsec, 2021-04-09:
@Shourai  ,the `watch_process` seems to be monitoring other processes and restart them when they occasionally fail. Therefore running it restarts other processes which are not intended.

U-Sus, 2021-04-10:
bid farewell to your prized possession yi-hack-v4. 
[https;//github.com/alienatedsec/yi-hack-v5](https;//github.com/alienatedsec/yi-hack-v5)

alienatedsec, 2021-04-10:
Please open a new issue if found in [this release](https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.3.0)

closed 2021-04-10 by alienatedsec