alienatedsec/yi-hack-v5 #3: Any chance for Yi Dome 1080p support?

t00, 2021-04-07:
@alienatedsec please consider building h20 image if possible, I will be happy to test. I tried using only home_y20 after renaming to home_h20, leaving original root_h20 from Keroi/yi-hack-v4 build - it did not work.

Can you possibly share some links to prepare own image - I am proficient with Linux, just going through the noise on Discord and having over 100 forks of v4 just overwhelms me to say at least.

Another question - should it be enough to just ssh contents of home partition to a running h20 image, replacing existing files? Maybe there is a way to mount home file using loop? I just need streaming support for home assistant with PTZ, ONVIF might do the trick.

MiSSHeaDShoT, 2021-04-07:
Issues I found on 1080 Dome:
1)MQTT settings: mqtt set to enabled , but upon restart is not started. Searches for a file in camera internal memory that is missing. Probably related to the issue that the hack must reside in SD card for now.  
2) I run MQTT manually but is not stable , sometimes registers movements ,sometimes not. I am not sure if there is some timeout in which the camera does not register movements at all or the library is the problem.
3)RTSP stream is very good and I confirm the bug @Antarctic74 mentioned that the streams are availble both regardless settings.
4)Very high load - 70% CPU . The default app rmm had this and not sure why. Also @t00 mentioned same.

What works:
1)RTSP as mentioned works very fast.
2)PTZ works.

alienatedsec, 2021-04-10:
The [latest release](https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.3.0) supports `yi_dome_1080p`

closed 2021-04-10 by alienatedsec