alienatedsec/yi-hack-v5 #4: Watermark issue

alienatedsec, 2021-04-08:
Hi @roleoroleo 

Would you be so kind to have a look at the watermark code as it seems to be corrupting the snapshot and the imggrabber binary.

[image]

I know it uses the watermark through this code
```
var APP = APP || {};

APP.snapshot = (function ($) {

    function init() {
        initPage();
    }

    function initPage() {
        jQuery.get('cgi-bin/snapshot.sh?res=high&base64=yes&watermark=yes', function(data) {
        image = document.getElementById('imgSnap');
        image.src = 'data:image/png;base64,' + data;
        image.style = 'width:100%;';
        })
    }

    return {
        init: init
    };

})(jQuery);
```

Also, the links functionality has the same logic and when removed, it works perfectly (highlighted):

[image]

[image]

roleoroleo, 2021-04-08:
Sorry, explain me better.
Is the imggrabber binary corrupted or the jpg image corrupted?

If the image is corrupted try to run it manually and check the log:
```
imggrabber -m model -r high -d > /tmp/img.jpg
```

alienatedsec, 2021-04-08:
That sounds like another migration issue, let me come back to you soon @roleoroleo and thanks for the hint

```
/tmp # imggrabber -m yi_home_1080p -r high -w -d > /tmp/img1.jpg
Debug on
Starting program
Resolution high
1617912648987 - mapping file /tmp/view, size 1586752, to b6c93000
1617912649009 - closing the file /tmp/view
1617912649013 - found latest frame: id 196, frame_counter 40587
1617912649016 - processing frame 196
1617912649040 - processing frame 196
1617912649061 - processing frame 196
1617912649080 - processing frame 196
1617912649100 - processing frame 196
1617912649121 - processing frame 197
1617912649141 - processing frame 197
1617912649161 - processing frame 197
1617912649190 - processing frame 198
1617912649214 - processing frame 198
1617912649240 - processing frame 199
1617912649261 - processing frame 199
1617912649280 - processing frame 200
1617912649301 - processing frame 200
1617912649320 - processing frame 201
1617912649341 - processing frame 201
1617912649361 - processing frame 201
1617912649380 - processing frame 202
1617912649401 - processing frame 202
1617912649421 - processing frame 202
1617912649441 - processing frame 203
1617912649463 - processing frame 203
1617912649491 - processing frame 204
1617912649515 - processing frame 204
1617912649541 - processing frame 205
1617912649542 - writing frame: frame_offset 1034092, frame_ptr b6d98dac, frame_length 37
1617912649570 - processing frame 206
1617912649570 - writing frame: frame_offset 1034136, frame_ptr b6d98dd8, frame_length 8
1617912649591 - processing frame 207
1617912649591 - writing frame: frame_offset 4, frame_ptr b6c9c644, frame_length 23545
7399610175785532 - frame found, exit loop
Decoding h264 frame
Adding watermark
get watermark /home/yi-hack/etc/wm_res/high/wm_540p_0.bmp error
water mark init error
Error adding watermark
```

roleoroleo, 2021-04-09:
I took a look at the error.
Simply imggrabber doesn't find the file /home/yi-hack/etc/wm_res/high/wm_540p_0.bmp
Check it it's present.

alienatedsec, 2021-04-09:
@roleoroleo 
Another issue occurred, now it is not path related

yi_dome (720p) only works with low resolution but high is killing the process. (see below). I am about to try yi_home.

Is there any additional way to troubleshoot?

```
/tmp/sd/yi-hack-v5 # imggrabber -m yi_dome -r high -w -d >> /tmp/1.jpg
Debug on
Starting program
Resolution high
1617964236839 - mapping file /tmp/view, size 654400, to b6df7000
1617964236842 - closing the file /tmp/view
1617964236845 - found latest frame: id 1, frame_counter 0
1617964236847 - processing frame 1
1617964236878 - processing frame 1
1617964236908 - processing frame 1
1617964236930 - processing frame 1
1617964236958 - processing frame 1
1617964236979 - processing frame 1
1617964237008 - processing frame 1
1617964237029 - processing frame 1
1617964237058 - processing frame 1
1617964237079 - processing frame 1
1617964237108 - processing frame 1
1617964237129 - processing frame 1
1617964237158 - processing frame 1
1617964237179 - processing frame 1
1617964237209 - processing frame 1
1617964237238 - processing frame 1
1617964237258 - processing frame 1
1617964237279 - processing frame 1
1617964237308 - processing frame 1
1617964237330 - processing frame 1
1617964237358 - processing frame 1
1617964237378 - processing frame 1
1617964237408 - processing frame 1
1617964237429 - processing frame 1
1617964237458 - processing frame 1
1617964237479 - processing frame 1
1617964237508 - processing frame 1
1617964237536 - processing frame 1
1617964237558 - processing frame 1
1617964237579 - processing frame 1
1617964237608 - processing frame 1
1617964237629 - processing frame 1
1617964237658 - processing frame 1
1617964237679 - processing frame 1
1617964237708 - processing frame 1
1617964237729 - processing frame 1
1617964237758 - processing frame 1
1617964237788 - processing frame 1
1617964237809 - processing frame 1
1617964237838 - processing frame 1
1617964237859 - processing frame 1
1617964237888 - processing frame 1
1617964237915 - processing frame 1
1617964237938 - processing frame 1
1617964237958 - processing frame 1
1617964237980 - processing frame 1
1617964238008 - processing frame 1
1617964238029 - processing frame 1
1617964238058 - processing frame 1
1617964238079 - processing frame 1
1617964238108 - processing frame 1
1617964238136 - processing frame 1
1617964238158 - processing frame 1
1617964238179 - processing frame 1
1617964238208 - processing frame 1
[... 79 lines cut ...]
1617964240209 - processing frame 6
1617964240229 - processing frame 7
1617964240259 - processing frame 7
1617964240287 - processing frame 8
1617964240309 - processing frame 8
1617964240339 - processing frame 9
1617964240359 - processing frame 9
1617964240379 - processing frame 10
1617964240409 - processing frame 10
1617964240438 - processing frame 11
1617964240459 - processing frame 11
1617964240489 - processing frame 12
1617964240509 - processing frame 12
1617964240530 - processing frame 13
1617964240559 - processing frame 13
1617964240581 - processing frame 14
1617964240581 - writing frame: frame_offset 249912, frame_ptr b6e3a478, frame_length 35
1617964240609 - processing frame 15
1617964240609 - writing frame: frame_offset 249952, frame_ptr b6e3a4a0, frame_length 8
1617964240639 - processing frame 16
1617964240639 - writing frame: frame_offset 249964, frame_ptr b6e3a4ac, frame_length 63456
7399678895262268 - frame found, exit loop
Decoding h264 frame
Killed
```

roleoroleo, 2021-04-09:
There is a know problem with snapshot.
The conversion process uses a lot of memory and these cams run at the limit.
Probably it runs out of memory when creates high res jpg.
Try to enable swap file.

alienatedsec, 2021-04-09:
@roleoroleo swap file does the trick, thanks again.

closed 2021-04-09 by alienatedsec