alienatedsec/yi-hack-v5 #5: Safe to update over yi-hack-v4 (using their update home/rootfs menu)?

tunerooster, 2021-04-09:
I am **SO** happy to see this release.  I have so many problems with rtspv4/viewd.  I have Allwinner on my Yi indoor cameras and it works flawlessly.  Did you base the rtsp server on the Allwinner code?

_Is it possible/safe to use the yi-hack-v4 web update (in the `Update` section) to upload and install the yi-hack-v5  `home` and `rootfs` files?_

I hope so, because access to my installed outdoor cameras is difficult, i.e. to retrieve the SD card...

Thanks for this release!  I am looking forward to using it with great anticipation!

alienatedsec, 2021-04-09:
> I am **SO** happy to see this release. I have so many problems with rtspv4/viewd. I have Allwinner on my Yi indoor cameras and it works flawlessly. Did you base the rtsp server on the Allwinner [...]

@tunerooster in theory, it will be possible; however, you need to ensure that the `yi-hack-v5` folder is in the root directory before uploading `rootfs` and `home` files.

Regardless, it isn't on my ToDo list but I might consider it in the future and will let you know whenever I had a chance to test.

tunerooster, 2021-04-10:
Thank you, but I am a little confused...

I uploaded home_h30 and rootfs_h30 via the upload menu of the v4 web interface.  Then I logged in (via ssh) and found the following:
```
# cd /tmp/sd
# ls -l
-rwxr-xr-x    1 root     root       7770752 Apr 10 11:55 home_h30
drwxr-xr-x    2 root     root         32768 Apr 10 11:52 log
drwxr-xr-x    4 root     root         32768 Apr 10 12:03 record
-rwxr-xr-x    1 root     root        970244 Apr 10 11:54 rootfs_h30
```
Is this where the files are supposed to be?  I'm not clear what you mean by "you need to ensure that the yi-hack-v5 folder is in the root directory".
Do you mean they should go in:
```
/tmp/sd/yi-hack-v5/home_h30
/tmp/sd/yi-hack-v5/rootfs_h30
```
I tried rebooting with the files in /tmp/sd and it bricked the camera.  I have attached the two log files which were created FYI.  I will await your advice before trying again...  Thanks very much for your patience and help!

[log_start_reset.tar.gz](https://github.com/alienatedsec/yi-hack-v5/files/6289812/log_start_reset.tar.gz)
[log_wifi_connected.tar.gz](https://github.com/alienatedsec/yi-hack-v5/files/6289813/log_wifi_connected.tar.gz)

alienatedsec, 2021-04-10:
@tunerooster 
```
/tmp/sd/home_h30
/tmp/sd/rootfs_h30
/tmp/sd/yi-hack-v5 (folder that has not been released yet)
```

> I tried rebooting with the files in /tmp/sd and it bricked the camera. I have attached the two log files which were created FYI. I will await your advice before trying again... Thanks very much for your patience and help!

It should not brick the camera but the hack would not work.

tunerooster, 2021-04-10:
By "brick" I just mean that I could no longer connect via the Yi App or the yi-hack.  I had to put yi-hack-v4 on the SD card and then it recovered.

I understand now, you are saying I need a folder/files that are not yet released.  I understood from your message above that I should be able to test the beta release???  Please clarify.

alienatedsec, 2021-04-10:
> By "brick" I just mean that I could no longer connect via the Yi App or the yi-hack. I had to put yi-hack-v4 on the SD card and then it recovered.
>
> I understand now, you are saying I need a [...]

Bear in mind i moved this issue from `yi-hack-v4` repo to `yi-hack-v5`; hence why the confusion. You can test some previous beta, with the same principle. 
```
/tmp/sd/home_h30
/tmp/sd/rootfs_h30
/tmp/sd/yi-hack-v4 (folder from the below release)
```
Some[ releases are here](https://github.com/alienatedsec/yi-hack-v4/releases)

tunerooster, 2021-04-10:
Yes, that is where I downloaded the `yi_outdoor_v0.2.3b.zip` , but it contains only the home and rootfs files.  I do not see the yi-hack-v4 folder, or v5.  Please again advise...

tunerooster, 2021-04-10:
My mistake!!!  I see I am wrong.  It does contain yi-hack-v4!  Sorry for the error.  I will proceed as you suggest!

alienatedsec, 2021-04-10:
https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.3.0

Follow this @tunerooster

wilb, 2021-04-10:
Attempting to upgrade a camera of mine currently running 0.2.4 of roleoroleo's Allwinner firmware. Still a fair bit of rubbish knocking about on the sdcard from when I installed & online upgraded it previously:

```
root@kitchen-camera:/tmp/sd# ls -l 
drwxr-xr-x    2 root     root         32768 Mar 21 21:10 Factory.done
drwxr-xr-x    2 root     root         32768 Mar 21 21:10 fw_upgrade
-rwxr-xr-x    1 root     root       6701200 Apr 10 14:54 home_y20
-rwxr-xr-x    1 root     root       3678073 Mar 21 21:10 home_y20gam.done
drwxr-xr-x    2 root     root         32768 Apr 10 17:38 log
drwxr-xr-x    3 root     root         32768 Mar 21 21:10 newhome
-rwxr-xr-x    1 root     root        844212 Apr  8 21:17 rRTSPServer
-rwxr-xr-x    1 root     root        844196 Mar 21 20:42 rRTSPServer.dist
drwxr-xr-x    2 root     root         32768 Apr 10 17:41 record
-rwxr-xr-x    1 root     root       1012312 Apr 10 14:54 rootfs_y20
-rwxr-xr-x    1 root     root      67108864 Apr  8 20:35 swapfile
drwxr-xr-x    9 root     root         32768 Apr 10 17:35 yi-hack-v5
-rwxr-xr-x    1 root     root      12604610 Apr 10 17:32 yi_home_1080p_0.3.0.tgz
``` 
Do I need to remove some of the content? home_y20gam.done & Factory.done maybe?

alienatedsec, 2021-04-10:
> Attempting to upgrade a camera of mine currently running 0.2.4 of roleoroleo's Allwinner firmware. Still a fair bit of rubbish knocking about on the sdcard from when I installed & online upgraded it previously:
> 
> 
> 
> ```
> 
> root@kitchen-camera:/tmp/sd# ls -l 
> 
> drwxr-xr-x    2 root     root         32768 Mar 21 21:10 Factory.done
> 
> drwxr-xr-x    2 root     root         32768 Mar 21 21:10 fw_upgrade
> 
> -rwxr-xr-x    1 root     root       6701200 Apr 10 14:54 home_y20
> 
> -rwxr-xr-x    1 root     root       3678073 Mar 21 21:10 home_y20gam.done
> 
> drwxr-xr-x    2 root     root         32768 Apr 10 17:38 log
> 
> drwxr-xr-x    3 root     root         32768 Mar 21 21:10 newhome
> 
> -rwxr-xr-x    1 root     root        844212 Apr  8 21:17 rRTSPServer
> 
> -rwxr-xr-x    1 root     root        844196 Mar 21 20:42 rRTSPServer.dist
> 
> drwxr-xr-x    2 root     root         32768 Apr 10 17:41 record
> 
> -rwxr-xr-x    1 root     root       1012312 Apr 10 14:54 rootfs_y20
> 
> -rwxr-xr-x    1 root     root      67108864 Apr  8 20:35 swapfile
> 
> drwxr-xr-x    9 root     root         32768 Apr 10 17:35 yi-hack-v5
> 
> -rwxr-xr-x    1 root     root      12604610 Apr 10 17:32 yi_home_1080p_0.3.0.tgz
> 
> ``` 
> 
> Rebooted but it didn't flash with the new firmware. Do I need to remove some of the content? home_y20gam.done & Factory.done maybe?

@wilb Allwinner is a different platform - not compatible, sorry.

wilb, 2021-04-10:
Aah apologies - I assumed because you'd credited roleoroleo & the image names were the same that they were compatible. Was only trying that before sacrificing my hard to reach outdoor one... About to attempt that though - currently running yi-hack-4 :crossed_fingers:

wilb, 2021-04-10:
:tada: just to confirm the upgrade from `TheCrypt0/yi-hack-v4` -> `alienatedsec/yi-hack-v5` worked fine by simply `scp`ing the extracted tarball to `/tmp/sd/` and rebooting the camera. Only minor issue is that the root password has been reset as part of the upgrade - any reminder as to what that is?

wilb, 2021-04-10:
Ah it's just reset it to an empty password - just an enter on the ssh prompt gets you in.

closed 2021-04-10 by alienatedsec

bonjour81, 2021-05-01:
Hi @wilb would you mind clarify what you scp-ing on /tmp/sd ?  only home_h30/rootfs_h30 or also the yi-hack-v5 folder ?
I tried upload them, v0.3.1 (the 3) by FTP, but without success.

alienatedsec, 2021-05-01:
@bonjour81 you need the folder as well

bonjour81, 2021-05-01:
Maybe I did something wrong earlier => I first tried to copy the 2 files on SD and it was kind of half working (telnet + ftp was ok). That's why I uploaded by ftp.

So  I restarted from scratch !  re-installed v4, then scp of all files to /tmp/sd  ( home_h30/rootfs_h30 + yi-hack-v5 folder, then unplug-replug => rock & roll! :+1:

PilaScat, 2021-05-19:
> Maybe I did something wrong earlier => I first tried to copy the 2 files on SD and it was kind of half working (telnet + ftp was ok). That's why I uploaded by ftp.
> 
> So I restarted from scratch ! re-installed v4, then scp of all files to /tmp/sd ( home_h30/rootfs_h30 + yi-hack-v5 folder, then unplug-replug => rock & roll! 👍
> thanks !

tried this but not working, yi outdoor with TheCrypt0/yi-hack-v4, copied with sd and scp, all three files, rebooted and rebooted but nothing

alienatedsec, 2021-05-19:
> tried this but not working, yi outdoor with TheCrypt0/yi-hack-v4, copied with sd and scp, all three files, rebooted and rebooted but nothing

@PilaScat the card filesystem needs to be FAT32

PilaScat, 2021-05-19:
@alienatedsec it is, tried multiple ways and times but nothing

alienatedsec, 2021-05-19:
@PilaScat check #46

ChrisWeiss, 2021-05-27:
I'm a bit confused about how this should work - I have a camera running the old v4 hack. I put the v5 files on a 4gb SD card and the camera boots off the card and runs the v5 firmware. If I take the card out, it falls back to the v4 firmware. Is there a way to move the new v5 stuff into the onboard flash, or is the process to always boot from the SD card?

alienatedsec, 2021-05-28:
@ChrisWeiss I am not sure how but there is no chance to revert to v4 if you remove your SD card. 

If you boot without SD or if you accidently delete the yi-hack-v5 folder, it will enable telnet and FTP on the next boot.

The home partition got lots of space, but the image decompression process is not designed to allow many files to be added to the onboard memory; hence, the combination of both and how v5 works.

ChrisWeiss, 2021-05-28:
That exactly describes what is happening. I mistakenly thought the
FTP/Telnet state was a leftover from the V4 install.

Since the card must be formatted to 4gb, does that mean that’s the largest
space available for local image storage?

alienatedsec, 2021-05-29:
> That exactly describes what is happening. I mistakenly thought the
> FTP/Telnet state was a leftover from the V4 install.
> 
> Since the card must be formatted to 4gb, does that mean that’s the largest
> space available for local image storage?
> 
> 
> > @ChrisWeiss <https://github.com/ChrisWeiss> I am not sure how but there
> > is no chance to revert to v4 if you remove your SD card.
> >
> > If you boot without SD or if you accidently delete the yi-hack-v5 folder,
> > it will enable telnet and FTP on the next boot.
> >
> > The home partition got lots of space, but the image decompression process
> > is not designed to allow many files to be added to the onboard memory;
> > hence, the combination of both and how v5 works.
> >
> >
> --- Chris

I will need to update this guide. You need a FAT32 partition to burn the firmware. That is 32GB MAX in Windows when you do the format. The guide should focus on creating a FAT32 partition on bigger than 32GB SD cards. Once the v5 loaded, use any size card and format it in the app, but ensure the yi-hack-v5 folder is on the SD card too.

ChrisWeiss, 2021-06-04:
So to be clear, you do not need a 4GB partition for the initial hack, just a FAT32 formatted card?

alienatedsec, 2021-06-05:
> So to be clear, you do not need a 4GB partition for the initial hack, just a FAT32 formatted card?

Yes