alienatedsec/yi-hack-v5 #10: [yi dome 1080p] MQTT seems not to work

majkers, 2021-04-11:
Hello,
I run Home Assistant with mosquitto. Seems that camrea tries to connect but then disconnects and motion detection is not working. This is part of mosquitto log:

`[12:02:20] INFO: [INFO] found mqtt_user on local database
1618135340: New client connected from 192.168.1.7 as kameraDome1 (p2, c1, k120, u'mqtt_user').
1618135571: Socket error on client kameraDome1, disconnecting.
1618136267: New connection from 192.168.1.7 on port 1883.
[12:17:48] INFO: [INFO] found mqtt_user on local database
1618136268: New client connected from 192.168.1.7 as kameraDome1 (p2, c1, k120, u'mqtt_user').
1618136269: Socket error on client kameraDome1, disconnecting.`

majkers, 2021-04-11:
@roleoroleo told me in issue from his hack to leave empty 'Topics Prefix for remote configuration' option to fix it. It worked for him but not here...

majkers, 2021-04-11:
Motion detection via MQTT is not working....

roleoroleo, 2021-04-11:
The problem logged by mqtt server is not related to mqttv4 and motion detection.
It's a login problem of another software module: mqtt-config.
Leaving empty 'Topics Prefix for remote configuration', this program exits.

majkers, 2021-04-11:
@roleoroleo maybe. User login and password are 100% OK. If ypou send some commands to run via SSH, I can do that. Maybe mqtt is not running at all or something...

majkers, 2021-04-11:
I do not see any mqttv4 that is present on yi-hack4 cameras

roleoroleo, 2021-04-11:
> Can't open mqueue /ipc_dispatch_1. Error: No such file or directory

Probably ipc_multiplexer is not running.

majkers, 2021-04-11:
It is not running and it is not in bin directory...

alienatedsec, 2021-04-11:
> > Can't open mqueue /ipc_dispatch_1. Error: No such file or directory
> 
> Probably ipc_multiplexer is not running.

It is not integrated. A subject for the next release

alienatedsec, 2021-04-13:
@majkers Please send me a message through discord if you would like to test MQTT before the next release

majkers, 2021-04-13:
hmmm it depends on when You are planning to release new version. Maybe for now I will wait for it :)

alienatedsec, 2021-04-13:
@majkers, I got it ready to test (you can see the commits above). Also, I fixed a discord link; thanks for letting me know.

majkers, 2021-04-14:
Ok. I might give it a try. No problem

tunerooster, 2021-04-17:
I have tested mqtt motion event on `0.3.1`.  It sort of works (i.e., the mqtt topic/message is published), but the load on the camera immediately start to rise dramatically, and the camera becomes unusable.  Here is a `top` on the camera (Yi Outdoor) from a bit after triggering motion and receiving the mqtt message:
```
Mem: 29924K used, 836K free, 1820K shrd, 40K buff, 3788K cached
CPU:  3.7% usr 95.0% sys  0.0% nic  0.0% idle  0.0% io  0.0% irq  1.2% sirq
Load average: 26.47 11.68 4.51 17/111 2267
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
  973     1 root     S    24616 79.9   0 40.2 ./rmm
 1590     1 root     R     8028 26.0   0 12.1 ./p2p_tnp
  793     1 root     S     4240 13.7   0  4.0 ./dispatch
 2246     1 root     R     1660  5.3   0  3.6 ./cloudAPI_real -c 306 -url https://api.us.xiaoyi.com/v5/alert/gen_presigned_url -uid TNPUSAK-270272-NCKUJ -keySec 8IVrA96o57YN6mjS -type 0 -sub_type 1
  550     2 root     RW       0  0.0   0  3.5 [mmcqd/1]
 1588     1 root     R     5188 16.8   0  3.1 ./mp4record
  199     2 root     RW       0  0.0   0  2.6 [kswapd0]
  785     2 root     RW       0  0.0   0  2.6 [RtmpMlmeTask]
 1683     1 root     D     2488  8.0   0  2.4 h264grabber -r high -m yi_outdoor -f
 1589     1 root     S     5340 17.3   0  2.0 ./cloud
  784     2 root     RW       0  0.0   0  2.0 [RtmpTimerTask]
 2245  2042 root     R     1488  4.8   0  1.9 top
 1593     1 root     R     2348  7.6   0  1.6 ./oss
    3     2 root     SW       0  0.0   0  0.9 [ksoftirqd/0]
  792     1 root     S     1044  3.3   0  0.8 ./log_server
  105     2 root     DW       0  0.0   0  0.7 [kusbotg]
 1676     1 root     D     2732  8.8   0  0.4 rRTSPServer -r high -p 554
```
The load appears to be caused by: `/tmp/sd/yi-hack-v5/bin/imggrabber`.
I tried setting `Topic Suffix for jpeg image` and `Topic Suffix for video list message` both to blank, but that did not do anything.  So I renamed `imggrabber` to `imggrabber.orig` and that seemed to work.

Is there a "proper" way to disable  the mqtt jpeg image being created and sent over mqtt?

Thanks...

roleoroleo, 2021-04-17:
I will check if there something strange in the code.
But in your top list, imggrabber is only 2.4 %
Probably the problem is another process.

tunerooster, 2021-04-18:
It may not be cpu.  It may be IO causing the load.  When I removed
imggrabber the problem went away.

alienatedsec, 2021-04-18:
This issue #23 could be related

roleoroleo, 2021-04-18:
I will add a semaphore tu run only one imggrabber.

tunerooster, 2021-04-18:
I do not need or want a snapshot published via mqtt every time motion is detected.  If the Topic suffix for jpeg is blank, what about ignoring the snapshot?  (I.e., in addition to fixing the load issue.)

roleoroleo, 2021-04-18:
Ok.

roleoroleo, 2021-04-18:
Done:
https://github.com/roleoroleo/mqttv4/commit/d97f5d21b8a0a918c00a7141a4c9bdedd90505a6

alienatedsec, 2021-04-19:
Now, both issues are resolved - the next release should be key to verify.

closed 2021-04-19 by alienatedsec

tunerooster, 2021-05-15:
> Yes, it's related to this issue.
> The new version contains 2 different services:
>
>    - speak.sh
>    - speaker.sh
>
> The first service converts a text into an audio file using an internal tts
> engine.
> The second service plays an audio file with the correct format.
>
> Check the wiki.

Ah, thanks!

This is only for the Allwinner platform, right?  It would be most useful
(to me anyway) on the Yi Outdoor camera (i.e., for my frontdoor camera) to
use when someone rings the doorbell.

For this, I assume I would have to lobby @alienatedsec/yi-hack-v5
***@***.***> to
implement this on the `Hi3518e` platform correct?  Would this be a
difficult thing for him to do?

All the best,
Dick

alienatedsec, 2021-05-16:
> Ah, thanks! This is only for the Allwinner platform, right? It would be most useful (to me anyway) on the Yi Outdoor camera (i.e., for my frontdoor camera) to use when someone rings the doorbell. [...]

@tunerooster you don't need to lobby anyone - the mqtt submodule is already linked to roleo's repo

alienatedsec, 2021-05-16:
> For this, I assume I would have to lobby @alienatedsec/yi-hack-v5 ***@***.***> to implement this on the `Hi3518e` platform correct?

Correct

>Would this be a difficult thing for him to do?

I am not sure yet