alienatedsec/yi-hack-v5 #12: A BIG Thank You!

tunerooster, 2021-04-11:
I just installed `yi-hack-v5 - v0.3.0 - BETA` on a `Yi Outdoor 1080p (9CUS)`.
I am happy to report that (after following the instructions this time) it installed and booted perfectly!

- RTSP came right up and displays clearly.
- The web interface is up and all the screens display properly.
- Motion recording is working.
- ssh is working.

_The load runs high (ie, in top): Load average: 7.03 7.82 6.82 1/99 9416
Is that expected?_

That is all I have checked right now.

I made a 256M swapfile on the SD card as follows (as root):
```
dd if=/dev/zero of=/tmp/sd/swapfile bs=1k count=256k
chmod 600 /tmp/sd/swapfile
mkswap /tmp/sd/swapfile
swapon /tmp/sd/swapfile
cat /proc/swaps
echo '/tmp/sd/swapfile  swap  swap  defaults'  >> /etc/fstab
```
**_For some reason, fstab is not mounting the swap file after a reboot._**
_Can anyone suggest how to get swap mounted after a reboot?_

I am **_very pleased_** and hopeful that after further use, this hack is as stable as Allwinner.  I will continue testing...

**_BIG thanks to alienatedsec!  You are a hero!_**

alienatedsec, 2021-04-11:
@tunerooster thanks for testing and coming back

Do you really need 256MB? You can find the Swap File in the config section which will make 64MB.

tunerooster, 2021-04-11:
**_No_**, I just came up with 256MB out of the air.  I looked for a config option, but obviously, I didn't look hard enough.  Thanks for enlightening me!

I will revert to the 64MB version available in the config and delete my version.  Thanks again for your help!

P.S.  Could you comment on the "Load average".  The load on my yi indoor cam (allwinner) is: `Load average: 1.06 0.77 0.78 1/128 26495`, but on the yi outdoor it is `Load average: 7.03 7.82 6.82 1/99`.  Is it just likely to be due to a slower processor?  ...just want to make sure it is not a problem.

alienatedsec, 2021-04-11:
> P.S. Could you comment on the "Load average". The load on my yi indoor cam (allwinner) is: `Load average: 1.06 0.77 0.78 1/128 26495`, but on the yi outdoor it is `Load average: 7.03 7.82 6.82 1/99`. Is it just likely to be due to a slower processor? ...just want to make sure it is not a problem. This is in "idle" mode... ie, no recording or rtsp clients etc.

Allwinner is a much quicker chipset; hence you will see a difference.

closed 2021-04-12 by alienatedsec