alienatedsec/yi-hack-v5 #13: [ Yi Home 1080p ] No recording being made - v0.3.0

Shourai, 2021-04-11:
Same as issue: https://github.com/alienatedsec/yi-hack-v5/issues/2

- Updated to 0.3.0
- Formatted the SDcard with the Yi app.
- Recordings are being made while the cloud functionality was still on.
- After turning off the cloud functionality (and turning on recording without cloud), no recordings are being made.

alienatedsec, 2021-04-11:
Thanks for testing @Shourai 

Has the `watch_process` enabled cloud?

Shourai, 2021-04-11:
Unfortunately I think it does.
After enabling `watch_process` it also enabled `p2p_tnp`.
Furthermore the `cloudAPI` is connecting to `https://api.eu.xiaoyi.com/v4/ipc/on_line`

alienatedsec, 2021-04-11:
> Furthermore the `cloudAPI` is connecting to `https://api.eu.xiaoyi.com/v4/ipc/on_line`

@Shourai this could be fixed in another way. Something for the next release

Shourai, 2021-04-11:
Oh I meant that `watch_process` also enables cloudAPI which then connects to xiaoyi.
So `watch_process` seems to enable _something_ that allows recording.
Or is it just necessary to have cloud enabled?

murphymorph, 2021-04-12:
I renamed files cloud, cloudAPI and cloudAPI_real and then started watch_process again. Then the records are available and no connections are made to the cloud.

alienatedsec, 2021-04-12:
@murphymorph renamed to what?

murphymorph, 2021-04-13:
I booted the cam with the cloud activated. Then rename it to `/home/app/cloud.bak` to keep a backup. Then the video are saved.
But I recognized that it doesn't work after a reboot.  :/

murphymorph, 2021-05-09:
I started to implement fake cloudAPI. With my Yi Outdoor it is working so far. The videos will be saved and no connections to the cloud are made (as far as I have seen). The script returns fake responses from the cloud.

Cloud anyone test the script?

Replace /home/app/cloudAPI_real with the attached one and the name "cloudAPI_real". At the moment the cloud feature have to enabled. 

[cloudAPI.txt](https://github.com/alienatedsec/yi-hack-v5/files/6447518/cloudAPI.txt)

alienatedsec, 2021-05-09:
@murphymorph thanks. I will commit and test soon.

alienatedsec, 2021-05-09:
> Replace /home/app/cloudAPI_real with the attached one and the name "cloudAPI_real". At the moment the cloud feature have to enabled.

@murphymorph I will try to incorporate your change to update [this code](https://github.com/alienatedsec/yi-hack-v5/blob/02c1ebe4aebe391c98db09d12e84d1a6584753ed/src/static/static/home/yi-hack-v5/script/cloudAPI#L20-L31), so if the no cloud settings, it would run your code. I think its better, and it would not require changing `cloudAPI_real`

What are your thoughts?

murphymorph, 2021-05-11:
As far as I have seen all calls to the cloud are made through the cloudAPI. So I don't think there is a need for `/etc/hosts`.

alienatedsec, 2021-05-12:
> As far as I have seen all calls to the cloud are made through the cloudAPI. So I don't think there is a need for `/etc/hosts`.

Calls are going through cloudAPI but if you block the domain at the host level (camera), it will not connect but still record(?)

I just need time to test but it is challenging at the moment

murphymorph, 2021-05-14:
If you block the domains at host level with `/etc/hosts`, the camera doesn't store the videos (at least in my case). `cloud` keeps calling `cloudAPI`, but nothing more happens.

l-Legacy-l, 2021-09-20:
Any news on that issue ?

alienatedsec, 2021-10-02:
Not at the moment - another build is likely to fix it

Rushmed, 2021-10-20:
Hi thanks for your work.
I am still not able to record videon on SD with having cloud disabled.
V0.3.2

bluehz, 2022-02-27:
Still watching this issue, my 27US is also crashing overnight but havent determined reason.

papagdog, 2022-04-05:
Is this project still in alive? Just curious. I have 2 yi outdoor 1080p 9CUS cams that won't record unless the cloud is enabled. I've read through the comments on a possible solution, but I'm not sure where to start. Any insight would be much appreciated.

alienatedsec, 2022-04-05:
@papagdog 

https://github.com/alienatedsec/yi-hack-v5/issues/88#issuecomment-1088595542

g1za, 2022-08-26:
Try to have a look here #164 
It worked for me.

alienatedsec, 2022-12-05:
fixed with `0.3.5` - please upgrade to `0.3.6`

closed 2022-12-05 by alienatedsec