alienatedsec/yi-hack-v5 #8: [yi dome 1080p] nothing displayed on Home page

majkers, 2021-04-11:
Hello besides links to stream nothing is being displayed on Home page:

[image]

alienatedsec, 2021-04-11:
@majkers Thanks for testing

If you run any of the below commands, can you get the output for each dropped over here?
`dd bs=1 count=4 skip=592 if=/tmp/mmap.info 2>/dev/null | cut -c1-4`
`dd bs=1 count=20 skip=656 if=/tmp/mmap.info 2>/dev/null | cut -c1-20`

majkers, 2021-04-11:
Second one gives some result:

`

65USXXXXRI0000
`

P.S. I changed a bit is this output in case it is some secret stuff of my camera ;)

alienatedsec, 2021-04-11:
Thanks, @majkers 

The same issue as with yi_cloud_dome_1080p.
I will need to amend `status.json` and `system.sh`. Let me get back to you later.

alienatedsec, 2021-04-11:
@majkers 
Try replacing the attached `status.json` from the `/tmp/sd/yi-hack-v5/www/cgi-bin` location.
[status.zip](https://github.com/alienatedsec/yi-hack-v5/files/6292085/status.zip)

alienatedsec, 2021-04-11:
@majkers please try the attached `status.json` and replace it from this location `/tmp/sd/yi-hack-v5/www/cgi-bin`
[status.zip](https://github.com/alienatedsec/yi-hack-v5/files/6292097/status.zip)

majkers, 2021-04-11:
Everything besides Hardware ID is displayed

MiSSHeaDShoT, 2021-04-12:
Hi, same problem here. Here is my output of those commands:
` dd bs=1 count=20 skip=656 if=/tmp/mmap.info 2>/dev/null | cut -c1-20`

alienatedsec, 2021-04-12:
This should fix the issue @majkers @MiSSHeaDShoT
- [status.zip](https://github.com/alienatedsec/yi-hack-v5/files/6299039/status.zip) and [service.zip](https://github.com/alienatedsec/yi-hack-v5/files/6299144/service.zip) extract and replace within `/tmp/sd/yi-hack-v5/www/cgi-bin`

- Also [system.zip](https://github.com/alienatedsec/yi-hack-v5/files/6299081/system.zip) extract and replace within `/tmp/sd/yi-hack-v5/script`

Restart the camera through WebUI

majkers, 2021-04-13:
Everything is being displayed together with Hardware ID. Thanks!

closed 2021-04-13 by alienatedsec