alienatedsec/yi-hack-v5 #15: [Yi Outdoor 1080p] High resolution snapshot segmentation fault

murphymorph, 2021-04-12:
Taking a high resolution snapshot does not working on a Yi Outdoor camera. Low resolution snapshot works.

Problem occurs with and without swap file.

```
/tmp/sd/yi-hack-v5 # imggrabber -m yi_outdoor -r high --debug
Debug on
Starting program
Resolution high
1618234144057 - mapping file /tmp/view, size 1586752, to b6d4d000
1618234144058 - closing the file /tmp/view
1618234144058 - found latest frame: id 143, frame_counter 13942
1618234144058 - processing frame 143
1618234144084 - processing frame 143
1618234144084 - writing frame: frame_offset 1022604, frame_ptr b6e500cc, frame_length 37
1618234144100 - processing frame 144
1618234144101 - writing frame: frame_offset 1022648, frame_ptr b6e500f8, frame_length 8
1618234144119 - processing frame 145
1618234144120 - writing frame: frame_offset 4, frame_ptr b6d56644, frame_length 59674
7400159931599476 - frame found, exit loop
Decoding h264 frame
Segmentation fault
```

alienatedsec, 2021-04-12:
@murphymorph there is a wrong resolution set - thanks for raising

alienatedsec, 2021-04-12:
@murphymorph Please extract [imggrabber.zip](https://github.com/alienatedsec/yi-hack-v5/files/6299586/imggrabber.zip) and replace within this location `/tmp/sd/yi-hack-v5/bin`

murphymorph, 2021-04-13:
The problem still exists. 

It's a bit strange. If I call `imggrabber` directly via ssh withtout `-m yi_outdoor` or with `-m yi_home_1080p`, the snapshot is taken correctly. I compared the settings `YI_OUTDOOR` in imggrabber.c with `YI_HOME_1080P`, but I can't see any difference.

Thanks for your work! I'm struggeling with the toolchain setup. I hope to get it running soon.

alienatedsec, 2021-04-13:
@murphymorph move toolchain to `/opt`

[image]

alienatedsec, 2021-04-13:
> The problem still exists.
>
> It's a bit strange. If I call `imggrabber` directly via ssh withtout `-m yi_outdoor` or with `-m yi_home_1080p`, the snapshot is taken correctly. I compared the settings [...]

@roleoroleo any ideas where the issue is?

roleoroleo, 2021-04-13:
I don't know, the code is ok.
@murphymorph
Did you test the last build?

murphymorph, 2021-04-13:
I tried imggrabber from 0.3.0 and the version from @alienatedsec above.

murphymorph, 2021-04-16:
I tested again and the problem is fixed with commit 1437195.

closed 2021-04-16 by murphymorph