alienatedsec/yi-hack-v5 #16: [Yi Outdoor 1080p] Testing to Confirm Compatibility

databoy2k, 2021-04-13:
I see I'm not the only one with an outdoor cam over here (thanks for doing this, by the way). I've installed it on a pair of cameras and it seems to be working.

What tests or information do you need from us to confirm compatibility?

tannd155, 2021-04-13:
Hi databoy2k,

It's good to hear that as I also have a Yi outdoor camera. I'm looking forward to a stable version of this hack to install on my camera.

Currently, I'm using the v4 hack for the camera and found the connection via RTSP is not stable. So I hope that the v5 with the swap file will help to solve this out.

majkers, 2021-04-13:
Could you please check MQTT?

alienatedsec, 2021-04-13:
Thanks for testing. Other than that, please report only new issues only. There are quite a few already.
@databoy2k

alienatedsec, 2021-04-25:
Closing as not an issue

closed 2021-04-25 by alienatedsec