alienatedsec/yi-hack-v5 #17: password

tunerooster, 2021-04-13:
After installing yi-hack-v5 **_OVER_** yi-hack-v4, I am unable to set the root password:
- Via the Config web page.
- Via running `passwd root` from the command line.
From the command line, running (as root):
```
passwd
New Passwd: mypw
passwd: warning: can't create backup copy '/etc/passwd-': Invalid cross-device link
passwd: Device or resource busy
passwd: can't update password file /etc/passwd
```
From the Config web page, entering a password simply did not take.  After saving and rebooting, the password remains null.

Please note, _this only fails when installing yi-hack-v5 over an existing yi-hack-v4 install_.  When I installed it on a new camera with no hacks, it worked.

Please advise if there are any work-arounds.

Thanks!

majkers, 2021-04-13:
In https://github.com/alienatedsec/yi-hack-v5/issues/9 I updated directly from theCrypt0 hack

tunerooster, 2021-04-13:
@majkers, what do you mean you "updated directly from theCrypt0 hack"?

Is there a work around from theCrypt0 v4?  If so, what did you do?  Thanks!

majkers, 2021-04-13:
I mean before applying yi-hack5 I had theCrypt0 original hack without @alienatedsec fork of it.

tunerooster, 2021-04-14:
Understood.  Me too.  Does changing the password work for you?

alienatedsec, 2021-04-25:
@tunerooster @majkers

`umount /etc/passwd` then you can run `passwd` to set a new password

I got a fix for the next release to make it happen through WebUI

commit 61fd2219 (2021-04-25): [Fix] Password fix #17 #9 & Attempt to fix booting #20 & FTP patch
	new file:   src/ftpd/ftpd.patch
	modified:   src/ftpd/init.ftpd
	modified:   src/static/static/home/yi-hack-v5/script/cloudAPI
	modified:   src/static/static/home/yi-hack-v5/script/system.sh
	modified:   src/static/static/rootfs/etc/init.d/S20yi-hack-v5 - Enables FTP and Telnet
	modified:   src/static/static/rootfs/etc/passwd
	modified:   src/static/static/rootfs/etc/profile
closed 2021-04-25 by alienatedsec