alienatedsec/yi-hack-v5 #18: Not publishing motion event to MQTT

tunerooster, 2021-04-13:
I have an Yi indoor camera running Allwinner that has been working (with MQTT) for a year or so.  I have setup MQTT on a Yi Outdoor exactly as it is on the Allwinner camera under V5.  Ie, MQTT is Enabled, server IP is the same, same port, Topic Prefix is set, Top Suffix is set, Motion Start/Stop etc...

However, when I trigger motion on the V5 camera, I get no MQTT message received from my monitor:
```
mosquitto_sub -h my-mqtt-host -t '#'
```
It appears the v5 release is not publishing anything when a motion event is triggered.

**I have tested both `mosquitto_sub` and `mosquitto_pub` on the V5 camera and they both work perfectly.**

Looking forward to a patch...

alienatedsec, 2021-04-13:
#10 already raised the same. Adding ipc_multiplexer should resolve your issue @tunerooster

alienatedsec, 2021-04-13:
Duplicate of #10

closed 2021-04-20 by alienatedsec