alienatedsec/yi-hack-v5 #23: Camera crashes when going from "off" state to "on" state

kwirk, 2021-04-15:
I had issues when camera goes from "off" to "on" state (via settings), causing camera to freeze. I found that numerous _imggrabber_ processes were being created and hanging, caused by calls from Home Assistant trying to update the current snapshot.

I suspect issue is all those processes waiting for the next frame, and then them all trying at once to pull the frame when the camera is turned on again.

As a work around (or maybe an acceptable solution?), I've changed `snapshot.sh` to return error JSON when camera config is in off state:
```diff
diff --git a/src/www/httpd/cgi-bin/snapshot.sh b/src/www/httpd/cgi-bin/snapshot.sh
index d7499da..aff8b2c 100755
--- a/src/www/httpd/cgi-bin/snapshot.sh
+++ b/src/www/httpd/cgi-bin/snapshot.sh
@@ -10,6 +10,17 @@ validateFile()
     esac
 }
 
+YI_HACK_PREFIX="/tmp/sd/yi-hack-v5"
+CONF_FILE="$YI_HACK_PREFIX/etc/camera.conf"
+
+if grep -q SWITCH_ON=no "$CONF_FILE"; then
+    printf "Content-type: application/json\r\n\r\n"
+    printf "{\n"
+    printf "\"%s\":\"%s\"\\n" "error" "true"
+    printf "}"
+    exit 0
+fi
+
 MODEL_SUFFIX=$(cat /home/app/.camver)
 BASE64="no"
 RES="-r high"
```

alienatedsec, 2021-04-16:
@kwirk Great suggestion, thanks.

A pull request will make it happen quicker :)

alienatedsec, 2021-04-17:
@kwirk Thank you. I will merge when back at home.

roleoroleo, 2021-04-18:
I will add a semaphore to run only one imggrabber session.

kwirk, 2021-04-18:
> I will add a semaphore to run only one imggrabber session.

If you have multiple clients polling for snapshots that may not be ideal, limiting it to just one. But maybe a sensible limit would be better (which may depend on camera model).

roleoroleo, 2021-04-18:
Another way: add a timeout using 2 threads.

kwirk, 2021-04-18:
> Another way: add a timeout using 2 threads.

Yeah, I think it makes sense to have a timeout.

closed 2021-04-19 by alienatedsec

roleoroleo, 2021-04-19:
I pushed a new commit.
https://github.com/alienatedsec/yi-hack-v5/commit/d6078814ffce40c13841cd5298c7613efc7ce77a
Sorry but it's not tested, I don't have a compatible cam.

PR #26 (2021-04-19): [Fix] Snapshot return service unavailable when camera off
Resolves #23
commit 29172c46 (2021-04-19): [Fix] Snapshot return service unavailable when camera off

Resolves #23