alienatedsec/yi-hack-v5 #32: Can't login to pure-ftpd on Yi Outdoor 1080p

tunerooster, 2021-04-18:
I am trying to use ftp to download recorded files (On yi-hack-v5 0.3.1 Yi Outdoor 1080p).  When I enter:
```
# lftp root:myrootpw@driveway
lftp root@driveway:~> cd /tmp/sd/record
cd: Login failed: 530 Login authentication failed
```
The same works on yi-hack 0.2.5 in my Yi Indoor camera.
(I am not using legacy ftpd on either camera).

On my Yi Indoor camera (yi-hack 0.2.5) I get the following:
```
# lftp root:myrootpw@garage
lftp root@garage:~> cd /tmp/sd/record
cd ok, cwd=/tmp/sd/record            
lftp root@garage:/tmp/sd/record> 
```
Have I forgotten something?  What might I be doing wrong?  Or is this a bug?

Thanks!

tunerooster, 2021-04-19:
If I use `lftp root@driveway` (i.e., without a password) it works.
I forgot the password updating was not working.

closed 2021-04-19 by tunerooster

alienatedsec, 2021-04-20:
@tunerooster when you run `passwd` in the console, you can use those credentials then.

tunerooster, 2021-04-21:
I tried that...  I got:
```
# passwd
Changing password for root
New password: 
Retype password: 
passwd: warning: can't create backup copy '/etc/passwd-': Invalid cross-device link
passwd: Device or resource busy
passwd: can't update password file /etc/passwd
```