alienatedsec/yi-hack-v5 #36: Camera Settings -> Switch on/off the camera

tunerooster, 2021-04-21:
When I turn the camera off using HTTP  page:
```
Camera Settings -> Switch on/off the camera -> OFF
```
It does _not_ turn the camera off.  E.g., I can still stream via rtsp.
Allwinner v5 works.

kwirk, 2021-04-21:
Working okay on Yi Dome 1080p.

Does `http://x.x.x.x:8080/cgi-bin/camera_settings.sh?switch_on=no` work?  This would avoid attempting to set other settings just in case they are the issue.

tunerooster, 2021-04-22:
Yes, this works:

```
http://root:xxxxxxx@driveway:8080/cgi-bin/camera_settings.sh?switch_on=no
http://root:xxxxxxx@driveway:8080/cgi-bin/camera_settings.sh?switch_on=yes
```

alienatedsec, 2022-09-27:
@tunerooster I cannot reproduce it in v0.3.3
Please check and/or close this issue if already resolved

closed 2022-12-05 by alienatedsec