alienatedsec/yi-hack-v5 #39: Can't get 0.3.0 to install

tunerooster, 2021-04-28:
I have successfully installed two  Yi Outdoor 1080p cameras with yi-hack-v5 v0.3.0.  One was on a clean SD card (which was not yet provisioned), and the other was **over** an existing yi-hack-v4 v0.2.0.  There were no problems encountered on either one.

I am now trying to install yi-hack-v5 on a third Yi Outdoor 1080p **over** an existing remote yi-hack-v4.  I have done everything exactly the same as the first two, i.e., from my PC:
```
# cd yi-hack-v5/yi-outdoor
# ls -l
total 8601
-rw-r--r-- 1 rwk rwk 7792172 Apr 14 05:03 home_h30
-rw-r--r-- 1 rwk rwk 1014184 Apr 14 05:03 rootfs_h30
drwxr-xr-x 9 rwk rwk      10 Apr 14 05:04 yi-hack-v5
# scp -rp ./ root@frontdoor:/tmp/sd/
home_h30                              100% 7610KB 218.8KB/s   00:34    
rootfs_h30                            100%  990KB 216.4KB/s   00:04    
mqttv4.conf                           100% 1854    53.5KB/s   00:00    
...
# ssh root@frontdoor
# reboot && exit
```
But when the camera comes back online, yi-hack-v4 has booted.  I rebooted again, but still it boots to yi-hack-v4.

_Is there some way to force it to install and boot yi-hack-v5?_

Thanks!

P.S.  The camera is in a difficult to access location, so I cannot readily retrieve the SD card.  I would like to get it to work remotely.

alienatedsec, 2021-04-28:
Hi @tunerooster Load another set of `rootfs_h30` and `home_h30` to your camera (e.g. yi-hack-v4) and then do the same with v5 set of files.

However, try the 'update' page of v4 (v5 works the same way) and upload both files.

tunerooster, 2021-04-29:
@alienatedsec I don't know if I am doing something wrong or what...

As suggested, using the yi-hack-v4 "Update" page, I uploaded v0.2.0 of v4:
```
./yi-hack-v4/yi_outdoor/home_h30
./yi-hack-v4/yi_outdoor/rootfs_h30
```
Reboot returned me 0.2.0 as expected.  I then did the same thing uploading v0.3.1 of v5:
```
./yi-hack-v5/yi_outdoor-0.3.1/home_h30
./yi-hack-v5/yi_outdoor-0.3.1/rootfs_h30
```
After reboot, it still did not upgrade... v4 rebooted.  I tried a third time uploading 0.1.1 of v4:
```
./yi_outdoor-0.1.1/home_h30
./yi_outdoor-0.1.1/rootfs_h30
```
Again, it still rebooted to v4 0.2.0.

How, exactly, does it know when to reinstall home and rootfs during a reboot?

Is there something else I can try?

I did not try updating home/rootfs using scp, as I assume the upload from the menu accomplished the same thing as using scp, correct?

Thanks so much for your help!

alienatedsec, 2021-04-29:
> As suggested, using the yi-hack-v4 "Update" page, I uploaded v0.2.0 of v4:
>
> ```
> ./yi-hack-v4/yi_outdoor/home_h30
> ./yi-hack-v4/yi_outdoor/rootfs_h30
> ```
>
> Reboot returned me 0.2.0 as expected. I then [...]

Are you sure the path is ok? I have a different output after uploading through the `update` function. Please ignore the size of `home_y18` and `rootfs_y18` as those are text files.

```
/tmp/sd # ls -ls
total 65728
    32 drwxr-xr-x    2 root     root         32768 Apr  9 13:28 System Volume Information
    32 -rwxr-xr-x    1 root     root             4 Apr 29 13:24 home_y18
    32 drwxr-xr-x    2 root     root         32768 Apr 29 12:14 log
    32 drwxr-xr-x   74 root     root         32768 Apr 29 12:22 record
    32 -rwxr-xr-x    1 root     root             6 Apr 29 13:24 rootfs_y18
 65536 -rwxr-xr-x    1 root     root      67108864 Apr 28 20:17 swapfile
    32 drwxr-xr-x    9 root     root         32768 Apr 29 13:11 yi-hack-v5
/tmp/sd #
```

tunerooster, 2021-04-29:
This is what I have (after uploading the last attempt, which was yi-hack-v4 v0.1.1):
```
root@frontdoor /tmp/sd # ls -lt
total 9216
drwxr-xr-x    2 root     root        131072 Apr 29 12:19 log
-rwxr-xr-x    1 root     root       7755640 Apr 29 12:17 home_h30
-rwxr-xr-x    1 root     root        967088 Apr 29 12:17 rootfs_h30
drwxr-xr-x    3 root     root        131072 Apr 28 11:52 save
-rwxr-xr-x    1 root     root             0 Apr 27 23:48 last-download
drwxr-xr-x  479 root     root        131072 Apr 27 23:48 record
drwxr-xr-x    9 root     root        131072 Apr 14 11:04 yi-hack-v5

```
Are you asking if I am sure that I am uploading to `/tmp/sd`?  Is so, the answer is yes.  As I am using the v4 Update Menu, I don't pick the directory to upload to, but as you can see from the `ls -l` the home/rootfs files are the most recent.  Is there something else I can check?  Would it be helpful if I posted the log files from the last boot attempt?

tunerooster, 2021-04-29:
You didn't say...  Do you know what triggers an install of home/rootfs?

alienatedsec, 2021-04-29:
> You didn't say... Do you know what triggers an install of home/rootfs?

Simple `Reboot`

tunerooster, 2021-04-29:
Yes, but it does not reinstall on every reboot, even though the home/rootfs files remain in the /tmp/sd directory???

tunerooster, 2021-04-29:
> Simple `Reboot`

I understand that a reboot triggers a reinstall, but my understanding is that only if a newer/older version of home/rootfs is uploaded into the /tmp/sd directory.  So something must check that, based on the current version.  So I am wondering if there is a file which contains the "current" version id, that is somehow not being honored, or that I could examine to troubleshoot the problem.

alienatedsec, 2021-04-29:
@tunerooster i haven't found anything in the code or base firmware as of yet

tunerooster, 2021-05-01:
I reformatted the SD card (using the `Yi App`), and uploaded a fresh copy of `yi-hack-v5`, but is still fails to flash the new `home_h30` and `rootfs_h30`.  It again boots to `yi-hack-v4`.

As I stated at the outset, I did this (i.e., the exact same procedure) on another camera, same make/model Yi Outdoor (also running yi-hack-v4) and it re-flashed and booted yi-hack-v5 flawlessly.

Would it be safe to delete the entire root filesystem (e.g., `rm -rf /` minus /tmp), while the system is running, and then reboot, with the only thing left in the system being yi-hack-v5 in /tmp/sd?  Would it be likely to solve the problem?

I don't want to brick the camera, or make it inaccessible remotely, as it is quite difficult to access in it's location.

alienatedsec, 2021-05-01:
@tunerooster I will get back to you when I am back at home. Check S20 script in '/etc/init.d', and you could try to amend it to the same content from ‘yi-hack-v5’

No guarantee of course.

tunerooster, 2021-05-01:
Thanks, I updated `S20yi-hack-v4` with the contents of `S20yi-hack-v5` and reboot, but it did not help.  I checked to be sure `/tmp/sd/yi-hack-v5/script/system.sh` exists as well.

Any further suggestions are greatly appreciated!

alienatedsec, 2021-05-01:
@tunerooster if you did that, it's likely not saved as it loads all of the hack services, even those from v4. I will be at home in about 25 minutes

tunerooster, 2021-05-01:
I just noticed, this...

I saved the old version, and created the updated version using vi and saved it.  But vi saves it without +x permission and I did not `chmod +x  S20yi-hack-v4`, so it may not have run after a reboot.  I just fixed that, and now it does appear to have started v5 services, although `rtspv4` is still running (not `rRTSPService`) probably because rtspv4 stole the port before rRSTPServer started.  Both web services are running, one on port 80 (v4) and the other on port 8080 (v5).

Please advise...

tunerooster, 2021-05-01:
Wait, I just realized that by saving the original in the init.d directory, it is likely running both (which is not likely what you intenede).  I am getting rid of the v4 version and rebooting...  Stay tuned...

tunerooster, 2021-05-01:
OK, I have removed the original `S20yi-hack-v4` and replaced it with the contents of the v5 version and rebooted.  I now have only one http server running, but `rtsbv4` is still running, and *not* `rRTSPServer`.  Not sure how or why...

tunerooster, 2021-05-01:
Ah, I see `S12rtsp` is starting `rtspv4`, so `rRTSPServer` fails because `rtspv4` is using the same port...

Where are you heading with this???

alienatedsec, 2021-05-01:
> Ah, I see `S12rtsp` is starting `rtspv4`, so `rRTSPServer` fails because `rtspv4` is using the same port...
> 
> 
> 
> Where are you heading with this???

You can delete S12 so it won't start

tunerooster, 2021-05-01:
OK, but are you saying this is the solution?

alienatedsec, 2021-05-01:
@tunerooster This is the workaround to your specific issue.

However, we have not considered any differences between cameras - I know you mentioned the same model and the same process but have you compared them all? e.g. SD cards and their sizes? I recon the card (lets say if the size is not supported) is not readable at the early stage of boot so it would not find `rootfs` and `home` files.

tunerooster, 2021-05-01:
I removed S12 and rebooted, but `rRTSPServer` is still not running, and now neither is `rtspv4`.

tunerooster, 2021-05-01:
YES, the SD card in this camera is 64G.  The SD card in the working camera is 16G!  Could that be the problem???

alienatedsec, 2021-05-01:
> YES, the SD card in this camera is 64G. The SD card in the working camera is 16G! Could that be the problem???

Most likely - I got 32GB (class 4 and class 10), and those are working fine. I haven't tried 64GB as of yet. This one is V30, so that might be a problem anyway.

tunerooster, 2021-05-01:
But it works on v4, and the bootloader flash worked on this card when I originally installed it.  The bootloader is Yi firmware, correct?  Is there a possible difference in this regard between v4 and v5?

tunerooster, 2021-05-01:
In any case, are you saying that once I resolve the rRTSPServer not starting issue, that I should be able to run fine with this setup.

Re the rRTSPServer issue, I just ran it manually and I get:
 ```
./rRTSPServer: can't load library 'libstdc++.so.6'
```

alienatedsec, 2021-05-01:
> But it works on v4, and the bootloader flash worked on this card when I originally installed it.  The bootloader is Yi firmware, correct?  Is there a possible difference in this regard between v4 and v5?
> 
> 
@tunerooster 
I could possibly use different versions required for the built (provided by Debian 9). I read some comments that TheCrypt0 was using Ubuntu or Debian 10.

tunerooster, 2021-05-01:
I think I can set my `LD_LIBRARY_PATH` to solve the rRTSPServer load error.  Can you tell me where I should do that (what file)?

tunerooster, 2021-05-01:
If you are fairly confident that it is the SD card, I may have to bite the bullet and replace the SD card with a 16G version.  Is that what you would do (at this point)?

tunerooster, 2021-05-01:
Hmm...  I can't find `libstdc` in `/tmp/sd/yi-hack-v5`:
```
find /tmp/sd/yi-hack-v5 -name 'libstdc*'
```
return nothing.   What am I missing?

alienatedsec, 2021-05-01:
Files :)

Give me one sec - I need to fire the VM

tunerooster, 2021-05-01:
rRTSPServer seems to be working, THANKS!   I guess I will try running with this config for a while...  I will let you know if anything else breaks.

Thanks so much for your help!!!  I will also let you know if the 16G SD card solves the problem, in case anyone else runs into this issue (if and when I can get access to the camera and swap it out).

Best regards!

tunerooster, 2021-05-01:
You can close this issue if you want.  If I run into other problems with this config, should I post here (even if closed) or open a new issue?

alienatedsec, 2021-05-01:
> You can close this issue if you want. If I run into other problems with this config, should I post here (even if closed) or open a new issue?
@tunerooster 
If any issue is related to the same config (running over `yi-hack-v4`, I will focus on flashing the `yi-hack-v5` properly first. As mentioned in the release notes, the `service` page will be used for upgrades in the future and no more manual files movements (at least, I hope).

As you have not flashed `rootfs`, some functions will not work as I have upgraded the `Busybox` and added some functions to support WebUI functionalities.

closed 2021-05-01 by alienatedsec

majkers, 2021-05-04:
Hmm having the same issue here. I am using 64GB sd card. @tunerooster can You give a step by step procedure how to upgrade from hackv4?

majkers, 2021-05-04:
And what if I use smaller in size sd card to flash and after it replace it with bigger (64GB) with yi-hack-v5 folder in it? Will it work? I have a difficult physical access to my camera so I need to be sure :)

tunerooster, 2021-05-05:
@majkers...
After you have uploaded the yi-hack-v5 files:
```
/tmp/sd/home_h30
/tmp/sd/rootfs_h30
/tmp/sd/yi-hack-v5
```
Upload the attached file to the camera (in /tmp/sd): [libstdc++.zip](https://github.com/alienatedsec/yi-hack-v5/files/6427133/libstdc%2B%2B.zip)
(_Note, this is really a _**tar**_ file, but github refuses to upload tar and zip does not support symlinks_, plus unzip does not appear to be part of yi-hack.)

ssh into your camera, then:
```
echo '/tmp/sd/yi-hack-v5/script/system.sh' > /etc/init.d/S20yi-hack-v4
cd /home/lib
tar xvf /tmp/sd/libstdc++.zip  # missing library required by rRTSPServer
reboot
```
That's it!  You should then see `rRTSPServer` running (**_not_** `rtspv4`).

Regards...

P.S. I don't know if you can reinstall the 64GB SD card after flashing from 16GB card.  I don't really see why not.  Maybe @alienatedsec has an opinion.  If you try it, let us know the outcome.