alienatedsec/yi-hack-v5 #41: Excessive access to raw.githubusercontent.com

worthdog, 2021-04-30:
Hi,

First off, great work on this project and thank you.

I wanted to highlight that pi-hole shows my camera accessing raw.githubusercontent.com every 15 seconds.
I assume this is checking for updates? or is it reporting back other data? 
If this is an update check then this seems excessive? Would there  be any benefit slowing this down?

alienatedsec, 2021-04-30:
Hi @worthdog

You couldn't ask about it at a better time. It checks for the latest version and you are right, it has a quick turnaround as I needed it for testing. I will amend it for the next release which I was planning to have yesterday 😄

alienatedsec, 2021-04-30:
@worthdog The code will keep trying connecting if it's blocked - once allowed it should be ok. However, you can stop it by editing [`system.conf`](https://github.com/alienatedsec/yi-hack-v5/blob/60c5213173bc228e02b7cf3d056733dc6000225b/src/static/static/home/yi-hack-v5/etc/system.conf#L35)

alienatedsec, 2022-12-06:
fixed an issue with crontab abuse - please update to `0.3.7`

closed 2022-12-06 by alienatedsec