alienatedsec/yi-hack-v5 #42: WiFi strength incorrect

ping-localhost, 2021-05-01:
I don't think anything should be able to have over 100% WiFi strength. 

# WiFi strength 
[image]

# SSH output:
```
/home/yi-hack-v5 # cat /proc/net/wireless
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000   82.  -59   -83        0      0      0      0      0        0     
 ```

# System details
[image]

alienatedsec, 2021-05-01:
@ping-localhost yes, I mentioned the same in the release. It looks like I could fix it easier with your output, but it won't be the same for other cameras.

murphymorph, 2021-05-12:
`cat /proc/net/wireless` returns different ranges on the cameras. 0-100 on YI Outdoor and I suspect 0-70 on others. On YI Outdoor `iwconfig` return a value of X/100 for the link quality. Probably on other cams there is a similar output like 70/70. I can't verify the output, because I only have a YI Outdoor.

Then the percentage could be calculate like this `iwconfig 2>&1 | grep Quality | sed -e 's/.*Link Quality=\([0-9]*\/[0-9]*\).*/\1/' | awk -F/ '{FS="/"}{print $1*100/$2;}'`

 https://github.com/alienatedsec/yi-hack-v5/blob/02c1ebe4aebe391c98db09d12e84d1a6584753ed/src/www/httpd/cgi-bin/status.json#L40

alienatedsec, 2021-05-12:
> `cat /proc/net/wireless` returns different ranges on the cameras. 0-100 on YI Outdoor and I suspect 0-70 on others. On YI Outdoor `iwconfig` return a value of X/100 for the link quality. Probably on [...]

Thanks @murphymorph I tried that option before you suggested and as much as it is an easy fix for some cameras, others will show 100% regardless of the signal strength or quality

```
          Link Quality=100/100  Signal level:-40 dBm  Noise level:-71 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

murphymorph, 2021-05-14:
ok... another try :) I found a simplified calculation based on the signal level.

`cat /proc/net/wireless  | awk 'END { if ($4 <= -100) { print 0 } else if ( $4 <= -50) { print (($4 + 100)*2) } else {print 100} }'`

I took this from [here on Stackoverflow](https://stackoverflow.com/questions/15797920/how-to-convert-wifi-signal-strength-from-quality-percent-to-rssi-dbm) This could be sufficient as a rough assessment of the signal quality.

ChenchaoZhao, 2021-07-23:
yi dome wifi strength is only 7%

g1za, 2022-08-06:
I moved to yi-hack-5 from 4.
142% wifi strength, Yi dome. 
Let me know if there is any command I can run on my camera (and how) to debug this.

alienatedsec, 2023-07-14:
Pull request #265 is still to be tested

alienatedsec, 2023-08-09:
After so many years and multiple attempts, I still cant get my head around it.
So, it works for a while and goes back to over 100% figures

[image]

young747, 2023-08-10:
Camera Details-

Firmware Version 	0.3.9
Baseline Firmware Version 	0.3.8
Base Version 	2.1.0.0E_201809191630
Model Suffix 	yi_home_1080p
Hardware ID 	48US

bug --- dispalyed 142% wifi strength, Yi Home 1080p

status.json (  /tmp/sd/yi-hack-v5/www/cgi-bin/status.json )
> WLAN_STRENGTH=$(cat /proc/net/wireless | awk 'END { if ($4 <= -100) { print 0 } else if ( $4 <= -50) { print (($4 + 100)*2) } else {print 100} }')

app.js  ( decompress  /tmp/sd/yi-hack-v5/www/js/all.js.gz )
> ;a=parseInt((parseInt(h.wlan_strength)*100)/70)

| level(dBm)  | claculated WLAN_STRENGTH( % ) | WebUI WiFi Strength_icon |
|--------|--------|--------|
| >= -50 | 100 | strong |
|-51 | 98 | strong |
|-52 | 96 | strong |
|... |  | strong |
|-60 | 80 | strong |
|... |  | medium |
|-70 | 60 | medium |
|... |  | medium |
|-80 | 40 | medium |
|... |  | weak |
|-90 | 20 | weak |
|... |  | weak |
|<= -100 | 0 | weak |

[How to fix]
Not needed calculation at the `app.js.gz`, because of calculated with WLAN_STRENGTH(%) at the `status.json`

- Edit app.js ( gzip -d app.js.gz )

replace
```
 ;a=parseInt((parseInt(h.wlan_strength)*100)/70
``` 
with
```
 ;a=parseInt(h.wlan_strength)
``` 

- Compress app.js  ( gzip app.js )
- Reboot
- Connect WebUI  ( Cache clear or Ctrl+F5 key)

Therefore, seen  WiFi Strength  from 0 to 100%

alienatedsec, 2023-08-10:
I can't believe I will say this - fixed with `0.4.0d` - thanks @young747 

If anyone could also check and confirm, I will close this issue after so many years 👍

g1za, 2023-08-11:
Just updated to v. 0.4.0 and so far (following a cache clean) the WiFi signal strength is show correctly (0-100 %).
Thanks both!

closed 2023-08-11 by alienatedsec