alienatedsec/yi-hack-v5 #43: yi outdoor 1080p - recording to sd card doesn't work

bednim, 2021-05-01:
hi!

I have tried fat32 64gb micro sd  and video is not recorded onto sd card.

camera: yi outdoor 1080p, 9cus, FW: yi-hack-v5 - v0.3.1 - BETA

thank you

alienatedsec, 2021-05-01:
> hi!
> 
> 
> 
> I have tried fat32 64gb micro sd  and video is not recorded onto sd card.
> 
> 
> 
> camera: yi outdoor 1080p, 9cus, FW: yi-hack-v5 - v0.3.1 - BETA
> 
> 
> 
> thank you

@bednim 
Possibly, the format in the yi app is needed

bednim, 2021-05-01:
But then it will remove the yi hack v5 folder from the card. Will it work
without it?

alienatedsec, 2021-05-01:
> But then it will remove the yi hack v5 folder from the card. Will it work
> without it?
> 
> 
> > hi!
> >
> > I have tried fat32 64gb micro sd and video is not recorded onto sd card.
> >
> > camera: yi outdoor 1080p, 9cus, FW: yi-hack-v5 - v0.3.1 - BETA
> >
> > thank you
> >
> > @bednim <https://github.com/bednim>
> > Possibly, the format in the yi app is needed

It will enable FTP if you reboot so you could upload the folder again.

alienatedsec, 2021-05-01:
@bednim  It will enable FTP if you reboot after formatting, and you can upload the folder again.

bednim, 2021-05-01:
> @bednim It will enable FTP if you reboot after formatting, and you can upload the folder again.

okay, that seems to work, is there a way to reboot the camera without the UI? or do I have to powercycle it?

thank you!

bednim, 2021-05-01:
ok - got it done via telnet.

thank you!

closed 2021-05-01 by bednim

aptonline, 2021-08-13:
Re-uploading the `yi-hack-v5` folder onto the root on the SD does not work for me.

Formatted the card in app as described above. Rebooted, connected via FTP and uploaded the folder, rebooted.  Web UI does not come back up.