alienatedsec/yi-hack-v5 #45: HA integration

roleoroleo, 2021-05-04:
I pushed a beta support for this hack.
Could anyone test it?

https://github.com/roleoroleo/yi-hack_ha_integration

ping-localhost, 2021-05-04:
Upgrading my yi-hack integration right now. Will report back.

ping-localhost, 2021-05-04:
> Upgrading my yi-hack integration right now. Will report back.

Seems to be working. Thank you very much. :)

louulouu, 2021-05-05:
Tried it with two of my Yi Dome 1080p, (yi-hack-v5 and HA Integration) and it seems to be working. I'll test it out and keep an eye on bugs. Thank you so much for all the efforts @roleoroleo and @alienatedsec! Awesome work!

thedeany, 2021-05-05:
The `status` entity always seems to say "Disconnected" for me even when I can access the cam through HA still. Additionally, sometimes on HA restarts, the integration seems to not be able to connect to the camera and then it just dies; HA never sees the camera again until another restart. The error in HA logs is:

```
Failed to get system configuration from device 192.168.x.x: error HTTPConnectionPool(host='192.168.x.x', port=80): Max retries exceeded with url: /cgi-bin/get_configs.sh?conf=system (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f9407c7dbe0>: Failed to establish a new connection: [Errno 111] Connection refused'))
```

thedeany, 2021-05-05:
Should be... the "WiFi Strength" on the cam's web interface homepage shows 135% 😅 and it's in the room adjacent to the room with my AP; only a single interior wall between the two.

roleoroleo, 2021-05-05:
I will check it.

thedeany, 2021-05-05:
It's almost as if the cam is a bit too slow on occasion to respond to the request, so it just times out. And sometimes it responds within that time and connects fine. Either way, it probably should have some sort of auto-reconnect ability I would think.

roleoroleo, 2021-05-05:
Please run the cgi manually.
http://IP-CAM:8080/cgi-bin/status.json

thedeany, 2021-05-05:
I get a response from that. I can always access the cam even when Home Assistant fails to. Do you need the output?

alienatedsec, 2021-05-05:
> The `status` entity always seems to say "Disconnected" for me even when I can access the cam through HA still. Additionally, sometimes on HA restarts, the integration seems to not be able to connect [...]

Is the port correct as I can see `port=80`?

thedeany, 2021-05-05:
Yes, I switched my HTTPD and ONVIF ports. Should I not have done that? I wasn't sure why that change was made, and I prefer being able to access the web interface without specifying the port so I swapped them. Was there a specific reason for the original swap that I missed?

alienatedsec, 2021-05-05:
@thedeany maybe worth testing with default ports and see if it makes any difference

ping-localhost, 2021-05-05:
I just checked my `yi-hack-MStar` camera (using port 8080) and it has the same issue, so I don't think this is a `yi-hack-v5` issue. 

[image]

roleoroleo, 2021-05-06:
> Do you need the output?

Yes, please.

roleoroleo, 2021-05-06:
> Yes, I switched my HTTPD and ONVIF ports. Should I not have done that? I wasn't sure why that change was made, and I prefer being able to access the web interface without specifying the port so I swapped them. Was there a specific reason for the original swap that I missed?

Historical reasons when the wsdd service wasn't there.

thedeany, 2021-05-06:
Output from `status.json`:

```
{
"hostname":"yi-hack-v5",
"fw_version":"0.3.0",
"home_version":"2.1.0.0E_201809191630",
"model_suffix":"yi_home_1080p",
"hardware_id":"48US",
"serial_number":"[REDACTED]",
"local_time":"Wed Apr 22 19:40:33 UTC 2020",
"uptime":"1358.77",
"load_avg":"5.38 4.61 3.62",
"total_memory":"30760",
"free_memory":"12148",
"free_sd":"100%",
"local_ip":"192.168.x.x",
"netmask":"255.255.255.0",
"gateway":"192.168.x.x",
"mac_addr":"[REDACTED]",
"wlan_essid":"[REDACTED]",
"wlan_strength":"95" 
}
```

Again, just to reiterate, sometimes it does connect so it seems like a timeout or race condition or something.

roleoroleo, 2021-05-06:
Try to update status.json to the last version.

thedeany, 2021-05-06:
Looks like updating to 0.3.1 works. The integration has survived several HA restarts. The `status` attribute still shows "Disconnected" though, even though everything is working. Not a huge deal for me though.

roleoroleo, 2021-05-07:
Please try the new code I pushed.
I changed timeout parameter from 5 to 10.

PatrickR68, 2021-05-07:
> I changed timeout parameter from 5 to 10.

Wouldn't it be better to make it configurable?

roleoroleo, 2021-05-07:
> Wouldn't it be better to make it configurable?

Yes.

ping-localhost, 2021-05-07:
> Please try the new code I pushed.
> I changed timeout parameter from 5 to 10.

Updated to the latest version, but it doesn't change anything for me (the RTSP camera works fine though).

[image]

[image].  [image]

roleoroleo, 2021-05-07:
About mstar, is the cam updated with the last version?

ping-localhost, 2021-05-07:
> About mstar, is the cam updated with the last version?

It is: 

[image]

[image]

roleoroleo, 2021-05-07:
When you say status, do you mean the binary_sensor or media_player?
If you mean binary_sensor you need to configure mqtt because the status is related to the birth/will messages sent from the cam through mqtt.
If you mean media_player the status comes from the cgi I described above and it should work.

ping-localhost, 2021-05-07:
@roleoroleo I meant the binary_sensor status, yes. Didn't know I actually had to turn on MQTT to use it, so I've done that now and it works for both mstar and v5. My apologies for wasting your time.

thedeany, 2021-05-07:
I was also unaware the status sensor was related specifically to MQTT. I can confirm with MQTT up and running, it correctly reports "Connected" now.

roleoroleo, 2021-05-07:
I will add a note in the readme.
Thanks.

alienatedsec, 2022-12-06:
I am going to refresh this issue.

@roleoroleo is this integration specific to any FFMPEG version? - I am asking mainly because of the below parameter
[image]

roleoroleo, 2022-12-06:
ffmpeg is used to convert the audio file when you play a media (for example for the tts engine) and as a backup method when a snapshot is called (if the webapi fails).
The parameter you highlighted is used in the 2nd case and it's add to the cmdline to force tcp connection.
I don't know if this parameter is compatible with all ffmpeg versions.

roleoroleo, 2022-12-06:
Reading the documentation, the option is supported.

alienatedsec, 2023-02-20:
#232 for reference