alienatedsec/yi-hack-v5 #46: [yi-outdoor] 64GB sd card does not updates from yi-hack-4

majkers, 2021-05-04:
Just like it is mentioned in https://github.com/alienatedsec/yi-hack-v5/issues/39 and I think we should take care of it. Seems that big sd cards like 64 GB fail to boot and load y-hack-v5 although yi-hack-v4 was working and booting just fine.

alienatedsec, 2021-05-04:
@majkers 
```
For microSD cards larger than 32 GB, Windows 10 only gives you the option to format as NTFS or exFAT. You can create a small partition (e.g 4 GB) on a large microSD card (e.g. 64 GB) to get the FAT32 formatting option.

insert microSD card into PC card reader
open Disk Management (e.g. Win+x, k)
Disk Management: delete all partitions on the microSD card
right click each partition > "Delete Volume..."
repeat until there are no partitions on the card
Disk Management: create a new FAT32 partition
right click on "Unallocated" > "New Simple Volume..."
Welcome to the New Simple Volume Wizard: click "Next"
Specify Volume Size: 4096 > "Next"
Assign Drive Letter or Path: (Any) > "Next"
Format Partition: Format this volume with the following settings:
File system: FAT32
Allocation unit size: Default
Volume label: Something
Perform a quick format: ☑
You should now have a FAT32 partition on your microSD card that will allow the camera to load the firmware files to update to yi-hack-v5.
```

`fdisk -l`

```
Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk0p1    0,0,17      1023,254,63         16   62333951   62333936 29.7G  c Win95 FAT32 (LBA)
/home/yi-hack-v5 #
```

alienatedsec, 2021-05-04:
You only need a FAT32 partition to flash the firmware to `0.3.1`, then you can use any card but please keep the `yi-hack-v5` folder.

I don't plan to add anything to `rootfs` or `home` partition in the future and the normal upgrading process through WebUI should be more than sufficient.

majkers, 2021-05-04:
It is formatted via yi app so I assume it is Fat32

alienatedsec, 2021-05-04:
@majkers run `fdisk -l` from the console

majkers, 2021-05-04:
Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk0p1    0,1,1       1023,254,63         63  125042687  125042625 59.6G 83 Linux

alienatedsec, 2021-05-04:
@majkers you need FAT32 to flash `rootfs` and `home` files onto internal memory. The same is stated on [`yi-hack-v4` page](https://github.com/TheCrypt0/yi-hack-v4#getting-started)

[image]

majkers, 2021-05-04:
so I really do not get it how it worked the first time I applied yi-hack-4. So what should I do with the rest of space left on card besides this fat32 partition?

alienatedsec, 2021-05-04:
@majkers 

> You only need a FAT32 partition to flash the firmware to `0.3.1`, then you can use any card but please keep the `yi-hack-v5` folder.
> 
> I don't plan to add anything to `rootfs` or `home` partition in the future and the normal upgrading process through WebUI should be more than sufficient.

majkers, 2021-05-05:
OK closing for now cause I have no time to physical access my camera.

closed 2021-05-05 by majkers