alienatedsec/yi-hack-v5 #47: Enabling MQTT causes RTSP to freeze on motion

reef-actor, 2021-05-05:
I have upgraded from v4 + rtsp to v5 0.3.1.
I get a nice reliable stream with MQTT off.  Switching on MQTT the stream stutters and grinds to a halt a few seconds after waving my hand in front of the camera.
My suspicion is it is caused by the resource usage of the motion snapshot feature which was not present in v4.
I have swapfile enabled.

I would appreciate some guidance in debugging this.  Is there a change I can make to disable the motion snapshot feature with MQTT enabled to test my theory?

alienatedsec, 2021-05-05:
@reef-actor please change the name of `imgrabber`
```
cd /tmp/sd/yi-hack-v5/bin/imggrabber 
mv imggrabber imggrabber_
```

reef-actor, 2021-05-06:
With imggrabber renamed it is working, operation continues and MQTT motion events are working 🎉.
I should mention that previously imggrabber was working through the snapshot links on index.html (both low and high resolution), so I wonder if imgrabber is being called with different arguments when it is a motion event?

roleoroleo, 2021-05-06:
mqttv4 doesn't use the correct model when it runs imggrabber.
I will fix it.

roleoroleo, 2021-05-07:
Maybe I fixed it.
https://github.com/alienatedsec/yi-hack-v5/commit/443ff17a25033906ef580e41a08974f32002b6a7
But I can't test it (no cam).

alienatedsec, 2021-05-07:
@roleoroleo Thanks. I have the weekend to test and to have another release, if so.

zsdd89, 2021-05-14:
Any updates regarding new release? With MQTT activated my outdoor cam is unusable, it crashes every time it detects a motion

roleoroleo, 2021-05-14:
> Any updates regarding new release? With MQTT activated my outdoor cam is unusable, it crashes every time it detects a motion

Leave blank the field motion_detection_image

tunerooster, 2021-05-14:
Perhaps this field should be blank by default.

reef-actor, 2021-05-18:
> Leave blank the field motion_detection_image

This works and allows the snapshot to be used, but it seems to stop the sending of motion_stop messages? I can't see anything in the code that could cause this, but it is resolved by reinstating the motion_detection_image field (and renaming imggrabber to prevent the crash).

Is anyone also seeing this?

ping-localhost, 2021-05-18:
> This works and allows the snapshot to be used, but it seems to stop the sending of motion_stop messages? I can't see anything in the code that could cause this, but it is resolved by reinstating the [...]

Can confirm, if I rename `imggrabber` to something else, my camera stays online and stable on a motion event. As soon as I change it back and motion happens everything grinds to a halt.

That said, leaving `Topic Suffix for jpeg image` empty is a decent enough solution for me (for now), since I use MotionEye for motion events (and the events are still stored on the camera as well).

ping-localhost, 2021-05-18:
> This works and allows the snapshot to be used, but it seems to stop the sending of motion_stop messages? I can't see anything in the code that could cause this, but it is resolved by reinstating the [...]

@reef-actor I can confirm, if I rename `imggrabber` to something else, my camera stays online and stable on a motion event. As soon as I change it back (and motion happens) my RTSP stream no longer works.

That said, leaving `Topic Suffix for jpeg image` empty is a decent enough solution for me (for now), since I use MotionEye for motion events (and the events are still stored on the camera as well).

roleoroleo, 2021-05-19:
Could you please run mqttv4 manually and post the output when a motion is detected?

reef-actor, 2021-08-09:
> Could you please run mqttv4 manually and post the output when a motion is detected?

Sorry for the delay

```home/yi-hack-v5 # kill 1655
/home/yi-hack-v5 # mqttv4
Starting mqttv4 v0.0.3
key: SOUND_DETECTION | value: sound
Unrecognized config.
Trying to connect... Client camera sending CONNECT
Client camera received CONNACK (0)
Client camera sending PUBLISH (d0, q1, r1, m1, 'camera/status', ... (6 bytes))
. connected!
Client camera received PUBACK (Mid: 1)
CALLBACK MOTION START
Client camera sending PUBLISH (d0, q1, r0, m2, 'camera/motion', ... (12 bytes))
Resolution high
Client camera received PUBACK (Mid: 2)
Client camera sending PINGREQ
Client camera received PINGRESP
Client camera sending PINGREQ
Client camera received PINGRESP
```

It looks like the motion stop callback doesn't get called

roleoroleo, 2021-08-11:
Reading the code, motion stop message is not related to the motion_detection_image field.
The only problem might be that mqttv4 crashes and doesn't manage the message.

reef-actor, 2021-08-12:
I think you're right, I can't see any connection between the two things. Is there a build that includes the commits you mentioned earlier with the imggrabber fix?

roleoroleo, 2021-08-18:
For a build you have to ask to alienatedsec.

alienatedsec, 2021-10-05:
#63 Also related

A new release by the end of this month

reef-actor, 2021-10-30:
I upgraded using the UI. After the upgrade I checked the swappiness value; it was 0.

alienatedsec, 2021-10-31:
> I upgraded using the UI. After the upgrade I checked the swappiness value; it was 0.

Weird, I checked each config once again, and there is no 0 value in any.

reef-actor, 2021-11-03:
It might just be something odd with my installation, I think it should be considered a separate issue anyway (if it happens for anyone else).

closed 2021-11-03 by reef-actor