alienatedsec/yi-hack-v5 #48: 47US cannot open web interface?

gene-code, 2021-05-06:
First of all, BIG THANK YOU for making a 100% free rtsp support firmware.
I actually don't mind paying usd5 for the yi-hack-v4, but my country banned to send
donation to the developer.

Anyways, after installed the firmware, all the LEDs are as described and finally a steady  blue
light appeared. By scanning my local wifi, I could see the camera is connected to 192.168.0.3
But when i open in browser, it just not opening any web interface. Any idea why?

I could connect with the Yi Home app, and it works normally. And could also confirm IP from
the app it is 192.168.0.3. Any idea why?

gene-code, 2021-05-06:
OMG. My bad.
I didn't copy the yi-hack-v5 folder with the firmwares.
Pls ignore this issue.

emirBytes, 2021-05-06:
Ok just additional notes.
For my Yi Home 47US, it seems that it requires the SD card to remain in the slot for this hack to work.
If i remove the sd card, suddenly it doesnt work again. Thought I add this to here for anyone who is facing similar issue.

alienatedsec, 2021-05-06:
> Ok just additional notes.
> For my Yi Home 47US, it seems that it requires the SD card to remain in the slot for this hack to work.
> If i remove the sd card, suddenly it doesnt work again. Thought I add this to here for anyone who is facing similar issue.

Yes, having the SD card with the `yi-hack-v5` folder is necessary. The camera will run `telnet` and `ftp` if the `yi-hack-v5` folder is not available, so you can upload it and restart it when necessary.

> I actually don't mind paying usd5 for the yi-hack-v4, but my country banned to send
> donation via paypal to the developer.

I also don't mind paying, and I did pay for my cameras some time ago, but we have two issues here which made me doing `yi-hack-v5`:
1. No further updates for `yi-hack-v4` where other projects (e.g. @roleoroleo) expanded a lot and brought new functionalities
2. I can see the idea of licensing is missing the OpenSource principles. I think donations should be gained through appreciation and rewarded rather than an idea for the little business. Besides, the transparency of the executables running on your cam is also affected as the additional binaries, and modules source codes are not visible to the end-user.

emirBytes, 2021-05-06:
> Yes, having the SD card with the `yi-hack-v5` folder is necessary. The camera will run `telnet` and `ftp` if the `yi-hack-v5` folder is not available, so you can upload it and restart it when necessary.

Ahh I see. So if I want to remove SD card, all I need to do is ftp to it and copy paste the yi-hack-v5 folder at root directory?

alienatedsec, 2021-05-06:
> Ahh I see. So if I want to remove SD card, all I need to do is ftp to it and copy paste the yi-hack-v5 folder at root directory?

@emirBytes That's correct

closed 2021-05-08 by alienatedsec