alienatedsec/yi-hack-v5 #49: App not working

hkrob, 2021-05-10:
1080p Dome user here, latest fw firmware installed (v0.3.1 - BETA)

I can't get the app to load the camera stream or connect at all - I've tried a bunch of different proxy chains settings.
Has anyone had luck with app/cloud access? Any tips?

alienatedsec, 2021-05-10:
@hkrob I will need a step-by-step explanation of what you tried so far and what is the exact model `xxCN` or `xxUS`?

Mr-Mentor, 2021-08-31:
Same problem, the version of thecrypt0 (v4) works fine for me, but it has blocked the RTSP Server (donation).

hkrob, 2021-08-31:
I've gone back to v4 actually, I paid for the license a while ago. I'd like
to use v5, so I hope development on it continues

Mr-Mentor, 2021-09-01:
> 
> I've gone back to v4 actually, I paid for the license a while ago. I'd like to use v5, so I hope development on it continues
> […](#)

So if you say that paid RSTP works for you, I think I'll do it too, thanks :).

hkrob, 2021-09-01:
Yes it works well for me.. but I've noticed people are having trouble
obtaining licenses now.. so not sure if the developer is still working on
pushing them out...

alienatedsec, 2022-12-05:
I am not sure if worth trying `0.3.6` but it would be easier for me if I had a Chinese version to test

alienatedsec, 2023-07-12:
@hkrob please try `9.3.8e` #263

alienatedsec, 2023-08-07:
@hkrob there was a problem with proxychain in general. I got it fixed with [this `0.4.0b` release](https://github.com/alienatedsec/yi-hack-v5/releases/tag/0.4.0b). Please, if you wish to try again, this would be great.

hkrob, 2023-08-07:
Hi @alienatedsec  you really don't give up do you? :)

I have installed 0.4.0b but still can't connect to the app (I re-paired etc)
Not sure if it's a problem with the proxychain config now? has the format changed?
What is the format you expect in proxychains.conf ?

alienatedsec, 2023-08-07:
@hkrob 
> you really don't give up do you? :)

Not easily - true :)

The format is also embedded in the `0.4.0b` through that code:

https://github.com/alienatedsec/yi-hack-v5/blob/a6453e4cd037b9e91bc0679c46c18b4fbbccf582/src/www/httpd/cgi-bin/set_configs.sh#L106-L110

The output provided in the `/?page=proxy` is added at the end of the below template.

https://github.com/alienatedsec/yi-hack-v5/blob/a6453e4cd037b9e91bc0679c46c18b4fbbccf582/src/proxychains-ng/proxychains.conf#L1-L8

hkrob, 2023-08-07:
Cool - got it, now trying to find China proxies that work!!

alienatedsec, 2023-08-07:
@hkrob 

Try using this string on the proxy config page. Press save and then test or try either way.

`socks4 61.171.50.169 6688;socks4 218.78.65.202 6688;socks5 1.180.0.242 7302;socks5 60.190.195.146 10800`

hkrob, 2023-08-08:
It looks like you've cracked the case!
I had to try a couple of times, but it eventually connected - I can now view via VLC and the App at the same time

Well done and thanks!

alienatedsec, 2023-08-09:
Fixed in `0.4.0c` onwards

closed 2023-08-09 by alienatedsec