alienatedsec/yi-hack-v5 #54: wifi disconnecting, wpa may crashed, now reset

k3s4, 2021-05-15:
Hello,

I have issue with constantly disconnecting WiFi from my Yi Outdoor camera -

[./rmm][5/13/22:52:26:887]: ai_get_aacstream: get aenc stream select time out
[./rmm][5/13/22:52:29:71]: ai_get_aacstream: get aenc stream select time out
[./dispatch][5/13/22:52:29:792]: wifi disconnected, now reconnect wifi
[./rmm][5/13/22:52:30:718]: motrk img is too old. storetm:1620946349-725901, diff:1009
[./mp4record][5/13/22:52:33:144]: msg snd success[./dispatch][5/13/22:52:33:145]: got rcd heartbeat
[./watch_process][5/13/22:52:34:512]: cloud crashed![./rmm][5/13/22:52:41:538]: got a new motion stop. send_motion_stop_msg
[./rmm][5/13/22:52:41:561]: msg snd success[./dispatch][5/13/22:52:42:632]: **wpa may crashed, now reset**
[./dispatch][5/13/22:52:42:766]: no usb dev

      Mode:Managed  Frequency=2.447 GHz  Access Point: 20:76:93:4E:A5:32
      Bit Rate=135 Mb/s
      RTS thr:off   Fragment thr:off
      Encryption key   Security mode:restricted   Security mode:open
      Link Quality=100/100  Signal level:-54 dBm  Noise level:-85 dBm
      Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
      Tx excessive retries:0  Invalid misc:0   Missed beacon:0
after reboot it works only for couple of hours..
anyone has similar issue and fixed that somehow?

reef-actor, 2021-05-17:
I see a motion event in the log, if you are using MQTT it may be related to issue #47 ?

k3s4, 2021-05-18:
I'm not 100% sure but it looks like after changing hostname in config to blank (previous set by me) in order to generate automatically and it looks like camera stops disconnecting from WiFi.. ;)

alienatedsec, 2021-05-18:
> I'm not 100% sure but it looks like after changing hostname in config to blank (previous I've set custom one) in order to generate automatically and it looks like camera stops disconnecting from WiFi.. ;)

@k3s4 
Changing it to blank would generate another hostname based on the MAC address? Is that correct? Or have you made it blank somewhere else and not WebGUI?

k3s4, 2021-05-18:
@alienatedsec blank in WebGUI - only chaning in WebGUI.

alienatedsec, 2021-05-18:
@k3s4 Has it changed the Hostname? to something around yi-7bc3

k3s4, 2021-05-18:
> @k3s4 Has it changed the Hostname? to something around yi-7bc3

yes

alienatedsec, 2021-05-18:
I will look into this; however, on another note, I just checked if leaving the old Busybox will change the camera's responses.  The original Busybox is at v1.20.2, and despite not supporting many functions and applets that are also buggy, I managed to run the hack.

```
BusyBox v1.20.2 (2016-01-25 13:29:17 CST) multi-call binary.
Copyright (C) 1998-2011 Erik Andersen, Rob Landley, Denys Vlasenko
and others. Licensed under GPLv2.
See source distribution for full notice.

Usage: busybox [function] [arguments]...
   or: busybox --list[-full]
   or: busybox --install [-s] [DIR]
   or: function [arguments]...

        BusyBox is a multi-call binary that combines many common Unix
        utilities into a single executable.  Most people will create a
        link to busybox for each function they wish to use and BusyBox
        will act like whatever it was invoked as.

Currently defined functions:
        add-shell, addgroup, adduser, arp, arping, ash, awk, bootchartd, cat,
        chmod, chown, chpasswd, cp, cryptpw, cttyhack, cut, date, dd, delgroup,
        deluser, depmod, df, du, echo, egrep, fdisk, fgrep, find,
        flash_eraseall, flashcp, free, fsync, getty, grep, halt, hd, hexdump,
        hush, ifconfig, init, insmod, iostat, kill, killall, killall5, linuxrc,
        ln, login, logname, ls, lsmod, lsof, lsusb, lzcat, lzma, md5sum, mesg,
        mkdir, mknod, mkpasswd, modinfo, modprobe, more, mount, mpstat, mv,
        netstat, passwd, ping, poweroff, ps, pwd, reboot, remove-shell, rm,
        rmdir, rmmod, route, sed, sh, sleep, su, sulogin, sync, sysctl, tail,
        tar, top, tr, udhcpc, umount, unlzma, usleep, vi, vlock, xargs
```

It was a dead-end to be fair

k3s4, 2021-05-28:
still the same issue..

alienatedsec, 2021-06-11:
Please check #70 as could be related

k3s4, 2021-06-13:
@alienatedsec - I don't think its related as my camera don't respond to ping requested etc it's completely offline until reboot - for now I have added reboot command to crontab everyday for kind od workaround..

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-06-16:
> Thinking about some script for checking wifi connection and then try reconnect

This could be useful, did you get a chance to write a script to do this?
I have a daily reboot on all cameras but my outdoor camera is crashing every hour if the stream is being viewed.

k3s4, 2021-06-17:
Looks like script works fine (after little modification):

`[Thu Jun 17 21:55:03 CEST 2021] Wifi OK
[Thu Jun 17 22:00:04 CEST 2021] Wifi OK
[Thu Jun 17 22:05:01 CEST 2021] CONNECTION KO. Reboot
[Thu Jun 17 01:10:00 CEST 2021] CONNECTION KO. Reboot
[Thu Jun 17 22:10:04 CEST 2021] Wifi OK
[Thu Jun 17 22:15:04 CEST 2021] Wifi OK
`

`now=$(date +"%Y_%m_%d")
LOGFILE=/tmp/sd/watchdog/logs/wd_log_$now.log

if ping -c4 -q 192.168.1.1 > /dev/null ; then
  echo '['$(date)'] Wifi OK' >> $LOGFILE
else
  echo '['$(date)'] CONNECTION KO. Reboot' >> $LOGFILE
  sync
  sync
  sync
  killall httpd
  sleep 1
  reboot -f
fi
`

I have removed interface down/up lines as this was not working (for whatever reason) and camera reconnect only after reboot.

alienatedsec, 2022-12-06:
if the issue persists, please update to `0.3.7` as I managed to fix many issues with crontab and other scripts
Closing as old and potentially no longer related to the latest version.

closed 2022-12-06 by alienatedsec