alienatedsec/yi-hack-v5 #56: Scroll search from app broken

eminusss, 2021-05-19:
In the app there's a red line at the bottom allowing you to scroll to different times of the day to see the events at that time. In version 3.0 this worked properly. After updating to 3.1, when you scroll to a previous time, it's snaps back to the present and does not show the saved video you scrolled to. 

See attached image of red timeline I mean. 

[image]

eminusss, 2021-05-19:
I should mention that I'm using the yi_home_1080p

alienatedsec, 2021-05-19:
@eminusss i had the same on v0.3.0. Please re-apply the record on motion settings in the settings and restart the camera

eminusss, 2021-05-20:
Doing that from the web interface did not fix it. But then I tried a slurry of resetting steps and one (or a combination of them) worked. I really wish I had documented the steps. 

Regardless, yes, some aspect of resetting did fix it. 

If you ever come across a way to replicate the error, I'll happily test it to make up for my lack of documenting.

alienatedsec, 2021-05-20:
> If you ever come across a way to replicate the error, I'll happily test it to make up for my lack of documenting.

@eminusss I will - Having the same model helps, but interestingly, it is the only model with this symptom.

closed 2021-05-20 by alienatedsec