alienatedsec/yi-hack-v5 #57: Cannot disable record to SD card

hkrob, 2021-05-24:
I want to completely disable writing to the SD card, but I can't get it to work...

> /tmp/sd/yi-hack-v5/etc # cat system.conf
> HTTPD=yes
> TELNETD=no
> SSHD=yes
> FTPD=no
> BUSYBOX_FTPD=no
> DISABLE_CLOUD=no
> REC_WITHOUT_CLOUD=no
> MQTT=no
> RTSP=yes
> RTSP_STREAM=high
> RTSP_AUDIO=no
> ONVIF=no
> ONVIF_WSDD=no
> ONVIF_PROFILE=high
> ONVIF_NETIF=wlan0
> ONVIF_WM_SNAPSHOT=no
> NTPD=yes
> NTP_SERVER=pool.ntp.org
> PROXYCHAINSNG=yes
> SWAP_FILE=yes
> RTSP_PORT=554
> ONVIF_PORT=80
> HTTPD_PORT=8080
> USERNAME=
> PASSWORD=
> TIMEZONE=HKT-8
> FREE_SPACE=25
> FTP_UPLOAD=no
> FTP_HOST=
> FTP_DIR=
> FTP_DIR_TREE=no
> FTP_USERNAME=
> FTP_PASSWORD=
> FTP_FILE_DELETE_AFTER_UPLOAD=yes
> CHECK_UPDATES=no
> SSH_PASSWORD=
> CRONTAB=

Version

> YI-HACK-V5
> v0.3.1 - BETA

System

> Firmware Version | 0.3.1
> Base Version | 1.9.2.0I_201812141405
> Model Suffix | yi_dome_1080p
> Hardware ID | 15CN
> 

Any idea?

alienatedsec, 2021-05-24:
```
killall watch_process
killall mp4record
```

@hkrob 

If that works you can add it at the end of `system.sh`

hkrob, 2021-05-25:
@alienatedsec that should be an option, I actually renamed mp4record and recbackup
(in the /home/app directory for anyone reading this..)

alienatedsec, 2021-05-25:
> @alienatedsec that should be an option, I actually renamed mp4record and recbackup
> 
> (in the /home/app directory for anyone reading this..)
> 
> 
> 
> 

Renaming it is likely to kick the watchdog module loaded into the kernel. That could make your camera not operational (e.g. constant reboot).

I don't recommend your workaround @hkrob

hkrob, 2021-05-25:
24 or so hours in, still working fine :)
But agreed it's a rough solution

alienatedsec, 2022-12-05:
`0.3.6` to fix many other and similar issues

closed 2022-12-05 by alienatedsec