alienatedsec/yi-hack-v5 #59: Yi Dome U Pro 2K

seventhplug, 2021-05-25:
I have an incoming camera Yi Dome U Pro 2K and i like the features stated here.. May i ask if this camera is supported? I hope you could include a picture of the supported cameras. Thank You

alienatedsec, 2021-05-25:
> I have an incoming camera Yi Dome U Pro 2K and i like the (Yi-Hack v5) features stated here.. May i ask if this camera is supported? I hope you could include a picture of the supported cameras. Thank You

@seventhplug Features like features can be added on any hack, but your camera is likely based on a different chipset to Hi3815v200 supported by this project.

The latest cameras are based on the Allwinner platform, so please have a look at those repos, e.g. from [Roleo](https://github.com/roleoroleo/yi-hack-Allwinner-v2)

seventhplug, 2021-05-25:
Thank you for pointing me to the right direction..

closed 2021-05-25 by alienatedsec