alienatedsec/yi-hack-v5 #60: New Yi Outdoor Not Flashing

tunerooster, 2021-05-25:
I just bought a new Yi Outdoor 1080p from Amazon.  I installed yi-hack-v5 on a 16GB SD card the same as I have on three other Yi Outdoor cameras.  However, this new one fails to flash.  It just boots Yi firmware.
- The camera is labeled (below the QR code): IFUSY59UOS
- On the label next to the QR code it says:  Model: YHS.3020
- Firmware is: 9.0.26.07_202104231123

The other cameras (that work) are labeled:

- (Below the QR code) 9CUSYFDJT7
- Model: YHS.3017

Is there anything I can try to get the new camera to work?  Or is there any other information needed to diagnose the problem?  Does anyone else have this camera or know anything about it (different chipset etc?)

I will have to return the camera if it cannot be flashed.    (very sad)

P.S. I also tried to flash yi-hack-v4 (also without success).

tunerooster, 2021-05-25:
Yikes!  What a dummy I am!  I failed to notice there is a separate release for IFUS!  I am closing this and will install that!  Sorry for my negligence!

closed 2021-05-25 by tunerooster

wolle2000, 2021-06-05:
Hello all, i also stupid. Where i can find the IFUS Release ? Thank for your great work!

brettvan1, 2021-06-05:
Hey Wolle, 

I’ve perplexed myself over trying to launch this firmware with the ifus yhs.3020 .  Every time i flash it, it fails to load up the main components of the firmware.  I have triple checked the release notes and my application of them.  I have determined that for this model of the camera, you have to use the reoreoleo all winner 2v for it to work on this model of the camera...

I’ve been successful with that... I think that it might be possible to upgrade to the v5 via ftp somehow after applying the allwinnerv2 version... that’s my theory but haven’t been able to test it yet

hope this helps

wolle2000, 2021-06-05:
Hello, Thanks for response. I tried also this version, but with this I didn't get onvif or ftp running. Only matt works for me, also direkt after reset to default. Do I something wrong.? onvif client says Everytime. Not response is empty.Greetings Wolle

alienatedsec, 2021-06-05:
> Hello, Thanks for response. I tried also this version, but with this I didn't get onvif or ftp running. Only matt works for me, also direkt after reset to default. Do I something wrong.? onvif [...]

V5 is not compatible with Allwinner. Please don't try mixing those two.

tunerooster, 2021-06-06:
IFUS YHS 3020 use:

https://github.com/roleoroleo/yi-hack-Allwinner-v2

wolle2000, 2021-06-06:
Thanks

https://github.com/roleoroleo/yi-hack-Allwinner-v2