alienatedsec/yi-hack-v5 #61: [Feature Request] Nextcloud / GDrive / Rsync Upload.

Minims, 2021-05-28:
Hi,

First I would like to thank you for this v5 ! It's really cool.

I saw I can already use FTP to upload videos on a server, but is it possible to add at least one a this option ? 

- Google Drive Upload : like this for example : https://github.com/roleoroleo/yi-hack-Allwinner.gdrive
- NextCloud Upload, via Webdav
- Or simply rsync  to a server with speed limiting.

This seems more secure to me than FTP. rsync would be the best for me I think. 

Thanks !

silly-questions, 2021-06-06:
The google drive upload option would be amazing if possible!

alienatedsec, 2021-06-11:
I might look into it later this month.

borodiliz, 2021-07-26:
@alienatedsec just having the `rsync` binary would be very appreciated. Thanks for this hack

n1ckh3ll, 2023-03-01:
Hi,
Any news on the GDrive update? 
Thanks!!!

outermedia-cbo, 2023-03-22:
Hi,

I just replaced all calls to `ftpput` in [ftppush.sh](https://github.com/alienatedsec/yi-hack-v5/blob/master/src/static/static/home/yi-hack-v5/script/ftppush.sh) with calls to `curl` via the Nextcloud WebDAV interface. 

It reuses the variables from the FTP configuration. 

Only problem is that the `curl` binary distributed with yi-hack-v5 does not support `https`.

```
/home/yi-hack-v5 # curl --version
curl 7.86.0-DEV (arm-hisiv300-linux-gnu) libcurl/7.86.0-DEV
Release-Date: [unreleased]
Protocols: dict file ftp gopher http imap mqtt pop3 rtsp smtp telnet tftp 
Features: alt-svc AsynchDNS IPv6 Largefile UnixSockets
```

[Nextcloud_WebDAV.patch](https://github.com/alienatedsec/yi-hack-v5/files/11040908/Nextcloud_WebDAV.patch.txt)

alienatedsec, 2023-03-22:
@outermedia-cbo I can include the library supporting https (I think I got that ready somewhere) or you could test with this version https://github.com/roleoroleo/yi-hack-utils/blob/main/v5/curl.tgz for now

nelutzu777, 2023-04-25:
Hi. I insist on this option and I will explain why. I have 4 YIcam. I created 4 Gdrive accounts and in each one a folder named Cam1, Cam2, Cam3, Cam4. Each IPcam records on the account/folder. Now I SHARE those 4 folders to my basic account. So I have access from my phone/PC to all the recordings, and Google automatically sorts them chronologically. The free and convenient method, easy to configure... I used this for years until Google made an update with Auth2.0 Error 401 
@outermedia-cbo @alienatedsec @roleoroleo @porunov  @avengermsoft @denven  Thank you for your time

denven, 2023-04-26:
> Hi. I insist on this option and I will explain why. I have 4 YIcam. I created 4 Gdrive accounts and in each one a folder named Cam1, Cam2, Cam3, Cam4. Each IPcam records on the account/folder. Now I [...]

Hi, @nelutzu777, I noticed you mentioned me twice to maintain gdrive code. 

- which gdrive uploader repo program are you using, 
- have you tried [yi-hack-MStar.gdrive](https://github.com/roleoroleo/yi-hack-MStar.gdrive)?
- and what are the camera folders and video file names like, since I only have one Yi camera used which has the following recorded file structure? are they the same as the following?

[image]

and if you are a onedrive user, you can also try [yihack-onedrive-uploader](https://github.com/denven/yihack-onedrive-uploader) which I am recently maintaining and works well.

nelutzu777, 2023-04-26:
> Hi, @nelutzu777, I noticed you mentioned me twice to maintain gdrive code.
>
> * which gdrive uploader repo program are you using,
> * have you tried [...]

I have 4 IPcams of this model:
Yi 1080p Home | rootfs_y20 | home_y20
I use [TheCrypt0/yi-hack-v4](https://github.com/TheCrypt0/yi-hack-v4) in tandem [avengermsoft/xiaomi_gdrive](https://github.com/avengermsoft/xiaomi_gdrive)  And everything worked perfectly until the moment when GOOGLE made changes with "Google OAuth2 changes of OOB causes drive API authorization failure".
I tried all gdrive_loader on github and none of them work. The authorization code does not work.
@avengernsoft  gdrive_loader works, but only for a few days and I get an invalid token error and it needs to be updated once every few days.

nelutzu777, 2023-04-26:
> Hi, @nelutzu777, I noticed you mentioned me twice to maintain gdrive code.
>
> * which gdrive uploader repo program are you using,
> * have you tried [...]

[image]
[image]
[image]
[image]

**I apologize for mentioning YIhack V4 a lot, but I would like to switch to YIhack V5 + GDrive_Loarer**

nelutzu777, 2023-04-26:
@denven 
/home/yi-hack-v5 # cd /tmp/sd/
/tmp/sd # sh yi-hack/gdrive/GDriveConf
Configuration is started. Wait for several seconds.
Enter your Client ID:
xxxxxxxxxxxxxxxxxxxxxxxxxxxxx.apps.googleusercontent.com
Enter your Client secret:
xxxxxxx-xxxxxxxxxxxxxxxxxxxxxxxxxxxx-RX

Put this link into your browser:
https://accounts.google.com/o/oauth2/v2/auth?scope=https://www.googleapis.com/auth/drive&state=state&redirect_uri=http://127.0.0.1&response_type=code&client_id=00000000000000-xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx.apps.googleusercontent.com&access_type=offline

Copy and paste your code here:
4/0AVHEtk5X-eXXXXXXXXXXXHbsi1UGQdF5D1ZS2kXXXXXXXXXXXXXXXXXn4N6WxJPMT03eh7I7l-g
/tmp/sd/yi-hack/bin/./curl: line 1: syntax error: unexpected word (expecting ")")
grep: /tmp/sd/yi-hack/gdrive/conf/response: No such file or directory
grep: /tmp/sd/yi-hack/gdrive/conf/response: No such file or directory
Your code was expired or it is not correct. Please refresh a link in your browser and enter a new valid code:

denven, 2023-05-01:
> @denven /home/yi-hack-v5 # cd /tmp/sd/ /tmp/sd # sh yi-hack/gdrive/GDriveConf Configuration is started. Wait for several seconds. Enter your Client ID: [...]

please check https://github.com/roleoroleo/yi-hack-MStar.gdrive/issues/14#issuecomment-1529491682

raffifu, 2023-06-01:
> [https://github.com/roleoroleo/yi-hack-utils/](https://github.com/roleoroleo/yi-hack-utils/blob/main/v5/curl.tgz)

Hi, @alienatedsec  i'm trying to use this binary but segmentation fault happen on Yi Home 47US, can you provide a guide to build module from scracth or provide curl binary that support https in  Yi Home 47US. I'm on my way to create script for uploading recorded video to telegram.

alienatedsec, 2023-06-01:
> segmentation fault happen on Yi Home 47US

@raffifu A screenshot or more details like which firmware is used could be beneficial

roleoroleo, 2024-01-24:
Sorry for the late reply.
Did you copy libopenssl libraries to /tmp/sd/yi-hack-v5/lib ?
Please pay attention to the symbolic links.
FAT32 doesn't support synlinks, so you have to copy the library 3 times with 3 names: *.so *so.X and *.so.X.Y

denven, 2024-01-26:
@roleoroleo to avoid the curl issue, probably getting a standalone curl program whose version is compitable with the hardware architecture is better than the one which has dependencies on the libs, and set the curl path into global env. I ran into the simmilar issues in the past, and I am using this [curl version](https://github.com/denven/yihack-onedrive-uploader/tree/75e5c3c3048557b6404179debd019ab3f665026e/bin)

denven, 2024-01-27:
> @denven it is all about how the app was built - it requires libs as those could be shared with other apps, which is to reduce the file size of such apps.

true, this makes sense.

so the thing is raffifu copied a builtin curl into sd card and ran into an error, and roleoroleo has already given the reason and answer.

if this is the case, I think my suggestion is also optional solution with less pain for the few popular used apps, worry-free about the size and dependencies on the sd card.

alienatedsec, 2024-01-27:
> if this is the case, I think my suggestion is also optional solution with less pain for the few popular used apps, worry-free about the size and dependencies on the sd card.

@denven That would be another issue as those devices have limited memory and CPU - loading apps with built-in libs is to load them to RAM, which is more concerning than almost unlimited SD storage.

cjd6568358, 2024-08-14:
I created a shell script to mount my cloud storage with Samba and overwrite the record directory on the SD card, which will cause the video to be recorded directly to the cloud storage space, avoiding SD card relay。
As of now, it's working great. At the same time I would like someone to integrate it (the Samba configuration) into the web management
[image]

https://github.com/alienatedsec/yi-hack-v5/discussions/395