alienatedsec/yi-hack-v5 #64: Changing from yi-hack-v4 from TheCrypt0 to V5

idanny76, 2021-05-30:
I installed the V4 from TheCrypt0 and I was trying to flash your V5. I have followed all the steps I have seen online, here and closed issues on it. I have not been successful in flashing your V5 over TheCrypt0 V4. Can I flash one firmware over the other?

idanny76, 2021-05-31:
I was searching on going back to stock and try V5, instead of trying to flash over the V4. I figured it out. The stock firmware after it is downloaded it has be renamed to a specific name. PDF file here shows names of files depending on the camera version. 

https://oss.yitechnology.com/images/support/homecam2usermanual.pdf

alienatedsec, 2021-06-05:
There are easy ways to install v5 over v4 and vice-versa. Going back to stock is a different process. The one described by the PDF is the firmware upgrade which is replacing some files but not the whole memory.

idanny76, 2021-06-06:
How do I go from the V4 from Crypt0to to your V5? I flash it, but it still shows the stuff from the tabs from V4 and white screen below without any options. I did brick one Yi dome camera 720p and is blinking  yellow constantly. Can you provide me the info to restore stick firmware and go from v4 to v5? Thank you.

From: alienatedsec
Sent: Saturday, June 5, 2021 6:39 PM
To: alienatedsec/yi-hack-v5
Cc: idanny76; Author
Subject: Re: [alienatedsec/yi-hack-v5] Changing from yi-hack-v4 from TheCrypt0to V5 (#64)

There are easy ways to install v5 over v4 and vice-versa. Going back to stock is a different process. The one described by the PDF is the firmware upgrade which is replacing some files but not the whole memory.

silly-questions, 2021-06-06:
I managed to go from V4 to V5 without a problem on 3 out of the 4 cameras I have.  Simply dropping the files and folders from the correct zip file, onto the SD card (and leaving them there after), and pluggin in the cameras shows v5 on the web server.  The 4th camera flashed ok, but the webserver stops loading pages and only shows the banner at the top of the page after a few minutes.

idanny76, 2021-06-07:
> I managed to go from V4 to V5 without a problem on 3 out of the 4 cameras I have. Simply dropping the files and folders from the correct zip file, onto the SD card (and leaving them there after), [...]

I have a the two Yi Dome 720p, about 5 of the Yi Dome 1080p, and one Yi Home 1080p, and Outdoor one. Some are at my house and other at my parents location. I had programmed the 2 Yi Dome 720p with the V4 of TheCrypt0to, when I found this V5 version with the free RTSP services. When I tried to flash the V5 over the V4... I had the same situation of the IP only showing the menu banner options, but the rest of the webpage was blank. 

I tried flashing the camera back to stock and just ended up bricking the camera by manually replacing files on FTP. I would like to see if I can unbrick it, but I have not been able to find a tutorial about it. I wanted to get it everything working on these two, before I start flashing the rest. I was hoping maybe someone could guide on the conversion and possible unbricking by flashing Stock Firmware back on.

idanny76, 2021-06-07:
I just tried the other 720p Dome that has the V4 from TheCryt0 and copied the the files from V4 yi_dome_0.3.1 folder into the root of the SD card and it will not flash. Camera just reboots and does not flash over the V4. Can anyone explain to me what I am doing wrong? I have the yi-hack-v5 folder, rootfs_v201 and home_v201 in the root directory of the SD card.

idanny76, 2021-06-07:
[image]

idanny76, 2021-06-07:
Figured it out..... I have been going to the cameras IP like I did for the V4 firmware and for the V5 you have to add port 8080. Hopefully some with a similar issue will get help from this post.

closed 2021-06-07 by idanny76