alienatedsec/yi-hack-v5 #65: PTZ for WEBRTC

onaggm, 2021-06-02:
Any idea how to add ptz function for webrtc for YI-DOME-1080 camera?

kwirk, 2021-06-02:
For [AlexxIT/WebRTC](https://github.com/AlexxIT/WebRTC)?

You should add the following to home assistant config first, and restart home assistant:
```yaml
rest_command:
  camera_ptz:
    url: 'http://{{ ip }}:8080/cgi-bin/ptz.sh?dir={{ dir }}'
```

And then to the WebRTC card config, replacing with IP address of your camera:
```yaml
ptz:
  service: rest_command.camera_ptz
  data_left:
    dir: left
    ip: 192.168.x.x
  data_right:
    dir: right
    ip: 192.168.x.x
  data_up:
    dir: up
    ip: 192.168.x.x
  data_down:
    dir: down
    ip: 192.168.x.x
```

onaggm, 2021-06-03:
Thanks! It works!

closed 2021-06-03 by onaggm