alienatedsec/yi-hack-v5 #67: [YI Outdoor 9CUS] Crashing

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-06-08:
-  Camera goes offline and stops displaying video within an hour after power-on.
-  Camera still responds to pings <5ms.
-  SSH and WebUI stop responding as well.
-  RTSP disabled, also tried (mv imggrabber imggrabber_).
-  Need to physically switch power off/on to recover.
-  Good USB power supply is being used (Samsung 2A).
-  yi-hack-v5 - v0.3.1 - BETA (upgraded from v4).

```
/home/yi-hack-v5 # mount
rootfs on / type rootfs (rw)
/dev/root on / type jffs2 (rw,relatime)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
tmpfs on /dev type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
/dev/mtdblock5 on /home type jffs2 (rw,relatime)
none on /dev/mqueue type mqueue (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime,size=16384k)
/dev/mmcblk0p1 on /tmp/sd type exfat (rw,relatime,fmask=0022,dmask=0022,iocharset=utf8,namecase=0,errors=remount-ro)
/dev/mmcblk0p1 on /tmp/sd/yi-hack-v5/www/record type exfat (rw,relatime,fmask=0022,dmask=0022,iocharset=utf8,namecase=0,errors=remount-ro)
```

```
/home/yi-hack-v5 # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.3M      1.2M     56.0K  96% /
tmpfs                    15.0M      4.0K     15.0M   0% /dev
/dev/mtdblock5           12.7M      8.9M      3.8M  70% /home
tmpfs                    16.0M      1.6M     14.4M  10% /tmp
/dev/mmcblk0p1           59.5G     24.5G     35.0G  41% /tmp/sd
/dev/mmcblk0p1           59.5G     24.5G     35.0G  41% /tmp/sd/yi-hack-v5/www/record
```

```
Mem: 29684K used, 1076K free, 1624K shrd, 2504K buff, 7300K cached
CPU: 50.2% usr 13.7% sys  0.0% nic 33.1% idle  0.0% io  0.0% irq  2.8% sirq
Load average: 6.27 3.24 1.28 2/100 2209
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
 1062     1 root     R<   24608 79.9   0 27.0 {ai_get_lpcmstre} ./rmm
 1064     1 root     S    24608 79.9   0  7.9 {motion_proc} ./rmm
 1048     1 root     S    24608 79.9   0  5.9 ./rmm
 2092     1 root     S    12772 41.4   0  3.2 ./p2p_tnp
 2091     1 root     S    12772 41.4   0  2.5 ./p2p_tnp
 2173  2166 root     R     1488  4.8   0  2.1 top -H
 1026     1 root     S    24608 79.9   0  1.9 {hi_Adec0_Send} ./rmm
  782     2 root     SW       0  0.0   0  1.9 [RtmpTimerTask]
 2093     1 root     S    12772 41.4   0  1.7 ./p2p_tnp
 1060     1 root     S    24608 79.9   0  1.5 {venc_get_h264st} ./rmm
 1863     1 root     S    12772 41.4   0  1.5 ./p2p_tnp
 1027     1 root     S    24608 79.9   0  1.2 {hi_Adec0_Dec} ./rmm
 1058     1 root     S<   24608 79.9   0  1.1 {ao_send_lpcmstr} ./rmm
 1061     1 root     S    24608 79.9   0  1.1 {venc_get_h264st} ./rmm
 2161  1626 root     S     1212  3.9   0  1.1 dropbear -R -B
 1552     1 root     S     4184 13.5   0  0.8 ./mp4record
 1063     1 root     S<   24608 79.9   0  0.7 {ai_get_aacstrea} ./rmm
  783     2 root     SW       0  0.0   0  0.7 [RtmpMlmeTask]
 2099     1 root     S    12772 41.4   0  0.2 ./p2p_tnp
  975     1 root     S     4236 13.7   0  0.2 ./dispatch
  974     1 root     S     4236 13.7   0  0.2 ./dispatch
  403     2 root     SW       0  0.0   0  0.2 [kworker/0:1]
  967     1 root     S    24608 79.9   0  0.1 ./rmm
 1554     1 root     S    12772 41.4   0  0.1 ./p2p_tnp
 2100     1 root     S    12772 41.4   0  0.1 ./p2p_tnp
 1165     1 root     S     1768  5.7   0  0.1 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-g
 1631     1 root     S     1004  3.2   0  0.1 ipc_multiplexer
    3     2 root     SW       0  0.0   0  0.1 [ksoftirqd/0]
 1057     1 root     S    24608 79.9   0  0.0 {msg_proc} ./rmm
 1059     1 root     S<   24608 79.9   0  0.0 {ao_get_711strea} ./rmm
 1016     1 root     S    24608 79.9   0  0.0 {hi_Aenc0_Get} ./rmm
 1864     1 root     S    12772 41.4   0  0.0 ./p2p_tnp
 1865     1 root     S    12772 41.4   0  0.0 ./p2p_tnp
 1862     1 root     S    12772 41.4   0  0.0 ./p2p_tnp
 1553     1 root     S     5332 17.3   0  0.0 ./cloud
 1889     1 root     S     5332 17.3   0  0.0 ./cloud
 1890     1 root     S     5332 17.3   0  0.0 ./cloud
 1887     1 root     S     5332 17.3   0  0.0 ./cloud
 1888     1 root     S     5332 17.3   0  0.0 ./cloud
  976     1 root     S     4236 13.7   0  0.0 ./dispatch
  791     1 root     S     4236 13.7   0  0.0 ./dispatch
 1562     1 root     S     4184 13.5   0  0.0 ./mp4record
 1556     1 root     S     2348  7.6   0  0.0 ./oss
 2070     1 root     S     1496  4.8   0  0.0 crond -c /tmp/sd/yi-hack-v5/etc/crontabs
 2166  2161 root     S     1492  4.8   0  0.0 -sh
 1664     1 root     S     1492  4.8   0  0.0 /sbin/udhcpc -i wlan0 -b -s /home/app/script/default.script -x hostname:ABC
 1677     1 root     S     1488  4.8   0  0.0 /usr/sbin/crond -c /var/spool/cron/crontabs/
    1     0 root     S     1480  4.8   0  0.0 init
 1596     1 root     S     1480  4.8   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 1626     1 root     S     1188  3.8   0  0.0 dropbear -R -B
  790     1 root     S     1044  3.3   0  0.0 ./log_server
 1557     1 root     S     1024  3.3   0  0.0 ./watch_process
  427     1 root     S<    1020  3.3   0  0.0 udevd --daemon
  472   427 root     S<    1020  3.3   0  0.0 udevd --daemon
 1783     1 root     S     1020  3.3   0  0.0 ./arp_test
  478   427 root     S<    1020  3.3   0  0.0 udevd --daemon
  550     2 root     SW       0  0.0   0  0.0 [mmcqd/1]
  199     2 root     SW       0  0.0   0  0.0 [kswapd0]
  677     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd5]
    5     2 root     SW       0  0.0   0  0.0 [kworker/u:0]
  413     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd4]
  105     2 root     DW       0  0.0   0  0.0 [kusbotg]
    7     2 root     SW       0  0.0   0  0.0 [kworker/u:1]
  103     2 root     SW       0  0.0   0  0.0 [khubd]
  710     2 root     SW       0  0.0   0  0.0 [flush-179:0]
 1993  1553 root     Z        0  0.0   0  0.0 [cloudAPI]
   88     2 root     SW       0  0.0   0  0.0 [sync_supers]
  335     2 root     SW       0  0.0   0  0.0 [mtdblock2]
  986     2 root     SW       0  0.0   0  0.0 [hidog]
  324     2 root     SW       0  0.0   0  0.0 [romblock1]
   90     2 root     SW       0  0.0   0  0.0 [bdi-default]
  340     2 root     SW       0  0.0   0  0.0 [romblock3]
  364     2 root     SW       0  0.0   0  0.0 [romblock6]
  367     2 root     SW       0  0.0   0  0.0 [mtdblock6]
  372     2 root     SW       0  0.0   0  0.0 [romblock7]
  375     2 root     SW       0  0.0   0  0.0 [mtdblock7]
  398     2 root     SW       0  0.0   0  0.0 [cfinteractive]
  348     2 root     SW       0  0.0   0  0.0 [romblock4]
  404     2 root     SW       0  0.0   0  0.0 [kworker/u:2]
  412     2 root     SW<      0  0.0   0  0.0 [deferwq]
  343     2 root     SW       0  0.0   0  0.0 [mtdblock3]
    2     0 root     SW       0  0.0   0  0.0 [kthreadd]
  709     2 root     SW       0  0.0   0  0.0 [flush-31:2]
    4     2 root     SW       0  0.0   0  0.0 [kworker/0:0]
  771     2 root     SW<      0  0.0   0  0.0 [cfg80211]
    6     2 root     SW<      0  0.0   0  0.0 [khelper]
  332     2 root     SW       0  0.0   0  0.0 [romblock2]
  784     2 root     SW       0  0.0   0  0.0 [RtmpCmdQTask]
  785     2 root     SW       0  0.0   0  0.0 [RtmpWscTask]
   91     2 root     SW<      0  0.0   0  0.0 [kintegrityd]
   93     2 root     SW<      0  0.0   0  0.0 [kblockd]
  351     2 root     SW       0  0.0   0  0.0 [mtdblock4]
  356     2 root     SW       0  0.0   0  0.0 [romblock5]
  359     2 root     SW       0  0.0   0  0.0 [mtdblock5]
  251     2 root     SW       0  0.0   0  0.0 [fsnotify_mark]
  264     2 root     SW<      0  0.0   0  0.0 [crypto]
  316     2 root     SW       0  0.0   0  0.0 [romblock0]
  319     2 root     SW       0  0.0   0  0.0 [mtdblock0]
 2170     2 root     SW       0  0.0   0  0.0 [flush-mtd-unmap]
  327     2 root     SW       0  0.0   0  0.0 [mtdblock1]
```

alienatedsec, 2021-06-11:
Please check #70 as could be related

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-06-11:
It also had the extra cron jobs and doesn't crash but the camera still has performance issues - it doesn't seem to crash anymore but it drops the video connection regularly and has high CPU usage, probably the same thing as #55.

alienatedsec, 2021-06-11:
> it drops the video connection regularly

@X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T That could be my fault as I updated some dependencies and reverted them back after the release. See d1ef3cb

alienatedsec, 2022-12-05:
Please upgrade to `0.3.6` to fix many other and similar issues

closed 2022-12-05 by alienatedsec