alienatedsec/yi-hack-v5 #70: [YI Dome 1080P 45US] Crashing

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-06-11:
-  Camera goes offline and stops displaying video within an hour after power-on.
-  Camera still responds to pings <5ms.
-  SSH and WebUI stop responding as well.
-  RTSP disabled.
-  Need to physically switch power off/on to recover.
-  Good USB power supply is being used (3AMP).
-  yi-hack-v5 - v0.3.1 - BETA (upgraded from v4).

```
/home/yi-hack-v5 # mount
rootfs on / type rootfs (rw)
/dev/root on / type jffs2 (rw,relatime)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
tmpfs on /dev type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
/dev/mtdblock5 on /home type jffs2 (rw,relatime)
none on /dev/mqueue type mqueue (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime,size=16384k)
/dev/mmcblk0p1 on /tmp/sd type exfat (rw,relatime,fmask=0022,dmask=0022,iocharset=utf8,namecase=0,errors=remount-ro)
/dev/mmcblk0p1 on /tmp/sd/yi-hack-v5/www/record type exfat (rw,relatime,fmask=0022,dmask=0022,iocharset=utf8,namecase=0,errors=remount-ro)
```

```
/home/yi-hack-v5 # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.3M      1.2M     84.0K  93% /
tmpfs                    15.0M      4.0K     15.0M   0% /dev
/dev/mtdblock5           12.7M      7.9M      4.8M  62% /home
tmpfs                    16.0M      1.6M     14.4M  10% /tmp
/dev/mmcblk0p1           59.5G     57.5G      2.0G  97% /tmp/sd
/dev/mmcblk0p1           59.5G     57.5G      2.0G  97% /tmp/sd/yi-hack-v5/www/record
```

```
Mem: 29260K used, 1500K free, 1660K shrd, 744K buff, 10452K cached
CPU: 68.7% usr 20.2% sys  0.0% nic  9.0% idle  0.0% io  0.0% irq  1.9% sirq
Load average: 7.24 4.28 1.85 2/97 3065
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
  991     1 root     R    24800 80.5   0 37.9 ./rmm
  993     1 root     R    24800 80.5   0 13.1 ./rmm
  972     1 root     S    24800 80.5   0 11.5 {hi_Adec0_Dec} ./rmm
  984     1 root     S    24800 80.5   0  9.2 ./rmm
 3028  2730 root     R     1488  4.8   0  2.3 top -H
  771     2 root     SW       0  0.0   0  2.1 [RtmpTimerTask]
  989     1 root     S    24800 80.5   0  1.8 ./rmm
  971     1 root     S    24800 80.5   0  1.6 {hi_Adec0_Send} ./rmm
 2682  1639 root     S     1212  3.9   0  1.2 dropbear -R -B
  990     1 root     S    24800 80.5   0  1.0 ./rmm
 1570     1 root     S     5200 16.8   0  0.9 ./oss
  987     1 root     S    24800 80.5   0  0.7 ./rmm
 1754     1 root     S     8032 26.0   0  0.7 ./p2p_tnp
 1566     1 root     S     5240 17.0   0  0.7 ./mp4record
  992     1 root     S    24800 80.5   0  0.5 ./rmm
  772     2 root     SW       0  0.0   0  0.5 [RtmpMlmeTask]
 1568     1 root     S     8032 26.0   0  0.3 ./p2p_tnp
  962     1 root     S     4240 13.7   0  0.3 ./dispatch
  961     1 root     S     4240 13.7   0  0.3 ./dispatch
  954     1 root     S    24800 80.5   0  0.1 ./rmm
 1758     1 root     S     8032 26.0   0  0.1 ./p2p_tnp
 1781     1 root     S     6352 20.6   0  0.1 ./cloud
  963     1 root     S     4240 13.7   0  0.1 ./dispatch
 1153     1 root     S     1764  5.7   0  0.1 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-g
  541     2 root     SW       0  0.0   0  0.1 [mmcqd/1]
    3     2 root     SW       0  0.0   0  0.1 [ksoftirqd/0]
   88     2 root     DW       0  0.0   0  0.1 [sync_supers]
  403     2 root     SW       0  0.0   0  0.1 [kworker/0:1]
  986     1 root     S    24800 80.5   0  0.0 ./rmm
  970     1 root     S    24800 80.5   0  0.0 {hi_Aenc0_Get} ./rmm
  988     1 root     S    24800 80.5   0  0.0 ./rmm
 1757     1 root     S     8032 26.0   0  0.0 ./p2p_tnp
 1755     1 root     S     8032 26.0   0  0.0 ./p2p_tnp
 1756     1 root     S     8032 26.0   0  0.0 ./p2p_tnp
 1753     1 root     S     8032 26.0   0  0.0 ./p2p_tnp
 1567     1 root     S     6352 20.6   0  0.0 ./cloud
 1779     1 root     S     6352 20.6   0  0.0 ./cloud
 1780     1 root     S     6352 20.6   0  0.0 ./cloud
 1777     1 root     S     6352 20.6   0  0.0 ./cloud
 1778     1 root     S     6352 20.6   0  0.0 ./cloud
 1577     1 root     S     5240 17.0   0  0.0 ./mp4record
  778     1 root     S     4240 13.7   0  0.0 ./dispatch
 2081     1 root     S     1496  4.8   0  0.0 crond -c /tmp/sd/yi-hack-v5/etc/crontabs
 1565     1 root     S     1492  4.8   0  0.0 /sbin/udhcpc -i wlan0 -b -s /home/app/script/default.script -x hostname:XYZ
 2730  2682 root     S     1488  4.8   0  0.0 -sh
 2795     1 root     S     1484  4.8   0  0.0 /bin/login --
 1687     1 root     S     1484  4.8   0  0.0 /usr/sbin/crond -c /var/spool/cron/crontabs/
    1     0 root     S     1480  4.8   0  0.0 init
 1604     1 root     S     1480  4.8   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack-v5/www/ -c /tmp/httpd.conf
 1769     1 root     S     1480  4.8   0  0.0 ntpd -p pool.ntp.org
 1639     1 root     S     1188  3.8   0  0.0 dropbear -R -B
  777     1 root     S     1044  3.3   0  0.0 ./log_server
 1571     1 root     S     1024  3.3   0  0.0 ./watch_process
  427     1 root     S<    1020  3.3   0  0.0 udevd --daemon
  467   427 root     S<    1020  3.3   0  0.0 udevd --daemon
 1829     1 root     S     1020  3.3   0  0.0 ./arp_test
  473   427 root     S<    1020  3.3   0  0.0 udevd --daemon
 1646     1 root     S     1004  3.2   0  0.0 ipc_multiplexer
  199     2 root     SW       0  0.0   0  0.0 [kswapd0]
  678     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd5]
  413     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd4]
  103     2 root     SW       0  0.0   0  0.0 [khubd]
  711     2 root     SW       0  0.0   0  0.0 [flush-179:0]
    7     2 root     SW       0  0.0   0  0.0 [kworker/u:1]
  105     2 root     DW       0  0.0   0  0.0 [kusbotg]
  335     2 root     SW       0  0.0   0  0.0 [mtdblock2]
  324     2 root     SW       0  0.0   0  0.0 [romblock1]
  327     2 root     SW       0  0.0   0  0.0 [mtdblock1]
  367     2 root     SW       0  0.0   0  0.0 [mtdblock6]
  340     2 root     SW       0  0.0   0  0.0 [romblock3]
   93     2 root     SW<      0  0.0   0  0.0 [kblockd]
    2     0 root     SW       0  0.0   0  0.0 [kthreadd]
  332     2 root     SW       0  0.0   0  0.0 [romblock2]
  404     2 root     SW       0  0.0   0  0.0 [kworker/u:2]
  375     2 root     SW       0  0.0   0  0.0 [mtdblock7]
  398     2 root     SW       0  0.0   0  0.0 [cfinteractive]
  348     2 root     SW       0  0.0   0  0.0 [romblock4]
  356     2 root     SW       0  0.0   0  0.0 [romblock5]
  412     2 root     SW<      0  0.0   0  0.0 [deferwq]
  343     2 root     SW       0  0.0   0  0.0 [mtdblock3]
  364     2 root     SW       0  0.0   0  0.0 [romblock6]
  710     2 root     SW       0  0.0   0  0.0 [flush-31:2]
  372     2 root     SW       0  0.0   0  0.0 [romblock7]
    4     2 root     SW       0  0.0   0  0.0 [kworker/0:0]
    6     2 root     SW<      0  0.0   0  0.0 [khelper]
  773     2 root     SW       0  0.0   0  0.0 [RtmpCmdQTask]
  774     2 root     SW       0  0.0   0  0.0 [RtmpWscTask]
   90     2 root     SW       0  0.0   0  0.0 [bdi-default]
   91     2 root     SW<      0  0.0   0  0.0 [kintegrityd]
 1702     2 root     SW       0  0.0   0  0.0 [flush-mtd-unmap]
  351     2 root     SW       0  0.0   0  0.0 [mtdblock4]
  999     2 root     SW       0  0.0   0  0.0 [hidog]
  359     2 root     SW       0  0.0   0  0.0 [mtdblock5]
  251     2 root     SW       0  0.0   0  0.0 [fsnotify_mark]
  264     2 root     SW<      0  0.0   0  0.0 [crypto]
  316     2 root     SW       0  0.0   0  0.0 [romblock0]
  319     2 root     SW       0  0.0   0  0.0 [mtdblock0]

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-06-11:
Could be down to this in system.sh (>> appending to the cron on each reboot).

```
if [ "$FREE_SPACE" != "0" ]; then
    echo "0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh $FREE_SPACE" >> /var/spool/cron/crontabs/root
fi
```

```
cat ../../var/spool/cron/crontabs/root
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh 5
```

alienatedsec, 2021-06-11:
I will have a limited time before 17 June to check this issue but you can comment it in the 'system.sh' so it will not kick off and will remain disabled.

```
#if [ "$FREE_SPACE" != "0" ]; then
#    echo "0 * * * * /tmp/sd/yi-hack-v5/script/clean_records.sh $FREE_SPACE" >> /var/spool/cron/crontabs/root
#fi
```

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-06-11:
Seems reliable after removing these jobs from the cron, but the outdoor camera is still having issues.

alienatedsec, 2021-06-14:
> Seems reliable after removing these jobs from the cron, but the outdoor camera is still having issues.

#63 should be helpful with swapfile

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-06-14:
> 
> 
> > Seems reliable after removing these jobs from the cron, but the outdoor camera is still having issues.
> 
> #63 should be helpful with swapfile

Edited init.sh with the following and rebooted but the video still drops, the camera also now doesn't respond to pings but can see it is still connected to my router.

`sysctl -w vm.dirty_background_ratio=2
sysctl -w vm.dirty_ratio=5
sysctl -w vm.dirty_writeback_centisecs=100
sysctl -w vm.dirty_expire_centisecs=500
sysctl -w vm.vfs_cache_pressure=200
sysctl -w vm.swappiness=40
echo 5 > /proc/sys/vm/laptop_mode
echo 60 > /proc/sys/vm/swappiness `

`/home/yi-hack-v5 # cat /proc/sys/vm/swappiness
60`

alienatedsec, 2021-06-14:
> Edited init.sh with the following and rebooted but the video still drops, the camera also now doesn't respond to pings but can see it is still connected to my router.
>
> ```
> sysctl -w [...]

This value means that it will start moving to the swapfile when the memory is 40% full. You might amend it to 20 to ensure you have the best performance. 60 works fine for `yi_dome` but that camera is limited with memory by design.

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-06-14:
> This value means that it will start moving to the swapfile when the memory is 40% full. You might amend it to 20 to ensure you have the best performance. 60 works fine for `yi_dome` but that camera is limited with memory by design.

It still crashes with a value of 20 unfortunately.

closed 2022-11-29 by alienatedsec