alienatedsec/yi-hack-v5 #72: [YI-Home 27US] Unable to connect to Mesh Network

ljlongwing, 2021-06-14:
I have 3 Yi cameras....one is a 22US, and the other two are 27US models.  At one point in the past, all 3 of them connected to my TP-Link Deco M5 Mesh network without issue.  At some point (I don't know when), the two 27US models are giving me errors connecting to it.  The DecoM5 is broadcasting on both 2.4 and 5GHz frequencies with the same SSID.  It's using WPA2-PSK [AES] as the encryption.

I also have an old Cisco E3000 on my network.

When I configure the old Cisco to broadcast on SSID Moose-2.4 (just so it's not the same as the Deco M5), I can connect both of them to that network without issue.  This whole time I figured it was just a problem with the Deco M5 not allowing proper authentication (even though everything else was working without issue.

Then today, I was looking at the ability to reconfigure the WiFi in the home screen and I noticed this

[image]

I know that this one is connected to the -2.4 one...so I assume the first entry in the list is just the currently connected one...but I could be wrong....the thing that caught me about it though is that it's got 3 'Moose' listed....my Deco M5 has 3 nodes to it, so I assume that those are the 3 individual nodes.

Is it possible that the firmware is in some way having trouble negotiating with 'one' of the nodes, and as such, can't connect to any of them?

I'm a programmer, but not exceedingly familiar with WiFi Mesh networks...so, I really don't fully know what I'm asking here, I'm just trying to figure out if there is something you would be able to do to help me out with this situation.

This is a situation for me because where I have one of the cameras, it can't reach my -2.4 network well and constantly drops out, but there is one of the mesh nodes in the room with it, so if it could use that one instead, I wouldn't have issues.

alienatedsec, 2021-06-15:
Were your mesh devices updated with new firmware recently? Or maybe around the time, it started having issues? The only thing that comes to my mind is the 27US driver would have issues with negotiating. Try changing the encryption on your mesh to a lower one and see how it goes. I understand that your mesh is advertising one SSID but with both 2.4 and 5Ghz(?) It is worth separating as some legacy devices (like YI cameras) will have an old incompatible driver not supporting it or having connectivity issues. For the same reason, I have several SSID’s around my house, and they all separated.

ljlongwing, 2021-06-21:
Sorry for the delayed response.  I tend to keep my devices up to date, so it looks like march was the last update on the router, not sure if that's when it started or not.  For now I've moved some things around my house and the network seems to be stable enough to support the camera....I was hoping you might be able to enable some sort of 'debug' logging on the camera to determine why it can't connect to the mesh routers.

silly-questions, 2021-07-10:
Have just had kind of the same issue crop up.  I also have 3 nodes to my mesh system that broadcasts one SSID for both 2.4 and 5Ghz.  Two of my three cameras are now refushing to connect out after a mesh restart.  Not had any new firmware updates etc, just the wifi restart.  I don't seem to have the option to split the bands sadly but notice when trying to change the wifi to a guest SSID I have set up, it also displays the SSID three times in the list on some.  The furthest away can only see one node of the mesh system, and that only displays the SSID once in the list.  Changing it to the guest wifi SSID has allowed it to connect ok.  Will see how long it connects for and will update the thread.  I agree, it would be very nice if possible to be able to determine which node is which when picking the wifi SSID.  Not sure how possible it is though.

Johnanater, 2022-03-06:
@ljlongwing , which hack did you use for your 22US camera? 

I found a version from yi-hack-v3 from the thread `https://github.com/shadow-1/yi-hack-v3/issues/25`, but it has issues with WiFi connectivity.

Thanks!

ljlongwing, 2022-03-07:
> @ljlongwing , which hack did you use for your 22US camera?
> 
> I found a version from yi-hack-v3 from the thread `https://github.com/shadow-1/yi-hack-v3/issues/25`, but it has issues with WiFi connectivity.
> 
> Thanks!

I used this hack, yi-hack-v5....which is why I posted my problem here on this repository.

closed 2023-07-17 by github-actions[bot]