alienatedsec/yi-hack-v5 #74: No audio ?

Maastodonte, 2021-06-18:
Hi, I'm from yi-hack-v4 and then new to the yi-hack-v5.

I flashed smoothlly the v5 on my Yi Cloud Dome 1080p.
After a quick tour I'm wondering if there is a way to get a separate audio input from the camera like in the v4 with a uri like /ch0_2.aac ?
It is quite handy with a baby :)

Thanks for your answers.

I won't mind either if there is already a flux with audio and video together ;)

pkishino, 2021-07-08:
yeah, just migrated and discovered NO audio..thats quite a bummer :/ I only use onvif/rtsp and not the yi app..connected via agentdvr to record

pkishino, 2021-07-08:
i tried setting rtsp_audio=yes in the system.conf but no difference

akbaryahya, 2021-09-06:
yes it only sounds through mi app and record files but when connecting via rtsp there is no sound, * use YI Outdoor Camera

MACscr, 2022-06-06:
Yikes, so this custom firmware doesn't do audio at all on rtsp streams? Obviously it would need to be a stream. No idea how two different streams would even be useful even if it was provided.

kapabel, 2022-06-29:
I've got the same problem. No sound :(

alienatedsec, 2022-12-05:
The next release `0.3.7(?)` will support audio over RTSP

MACscr, 2023-01-01:
> 

So i guess still no audio? the 0.3.7 beta notes are a bit confusing. Says audio is next, so its not included?

alienatedsec, 2023-01-03:
> So i guess still no audio? the 0.3.7 beta notes are a bit confusing. Says audio is next in the 0.3.7 notes, so its not included as you had previously mentioned?

Unfortunately, I had to release another version with no audio, but the RTSP sound is already in the code. However, there are problems with HA if Audio is enabled as the platform does not support the audio format. I will think if it is worth releasing this way

MACscr, 2023-01-07:
What is the audio format? Why isn’t it in an rtsp format that the generic camera feature of HA can use?- Mark

Loafdude, 2023-03-31:
Is it possible to enable audio in the current build through configuration or does it require a new release?

alienatedsec, 2023-03-31:
@Loafdude new release - the code is right there

Loafdude, 2023-04-02:
Yes I see the code is in master.
Is there a guide to setting up a dev environment anywhere? Or could someone compile current master for y20?

Loafdude, 2023-04-05:
OK I was able to compile. (Yikes that SDK needs a lot of patching!)

Enabling audio seems to break both the low and high quality streams on my builds.
I've disabled both cloud, onvif. Only things enabled are SSH, RTSP and Audio.

VLC begins to play back then I loose audio and video becomes very laggy.
VLC Statistics show:
[image]

Disabling audio plays back 100% normally.

It could be my build. Perhaps someone can confirm? 
I'm using a Home 1080p.

The h264grabber -r audio is only using 1.3% cpu. 
Nothing seems out of line there.

Perhaps the audio stream is malformed and VLC is choking on it?

I understand it's experimental. Just trying to help out!

Loafdude, 2023-04-05:
Oh I forgot to mention that the audio only stream appears to work fine

alienatedsec, 2023-04-05:
@Loafdude @Maastodonte I need a couple more days to release another version with audio support.

Loafdude, 2023-04-06:
No problem, let me know if I can be helpful. I will test once available.

closed 2023-08-20 by github-actions[bot]