alienatedsec/yi-hack-v5 #75: Setting a static IP

Maastodonte, 2021-06-18:
Hi guys,

I was wondering how we can set a static IP as there is no option on the web interface.

I tried to do the following
`ifconfig eth0 down; ifconfig eth0 192.168.x.x netmask 255.255.255.0; route add default gw 192.168.x.x eth0; ifconfig eth0 up;`
It works but on restart it goes with DHCP again...

I looked on the filesystem but didn't find an interfaces file or anything talking about DHCP or network parameters. Only some wpa_supplicant files.
Do I have to create /etc/network/interfaces myself ? Will it work ?

Please share the best way to deactivate DHCP and set a static IP on yi-hack-v5.

Thanks.

untiltheendoftimes, 2021-06-26:
You can do it the other way around. Set a dhcp bind from your router to the camera.

Maastodonte, 2021-07-07:
That's what I ended up doing but I had to connect to my router by telnet and find the right commands.

It would have been nice if the camera natively provided such feature.

Thanks anyway ;)

closed 2021-07-07 by Maastodonte