alienatedsec/yi-hack-v5 #76: YI Outdoor Camera 1080p YHS.3020 - Hack cannot be installed

Johanson2020, 2021-06-19:
Hello I'm trying to flash my factory new camera with the Yi- hack v5.
SD card 16Gb FAT32
i copied the home_30 & the rootfs_30 to the top level of the sd card.
SD card inserted but the camera start normally without installing the firmware....
Does anyone have any idea?
Thanks a lot!
Johannes

Johanson2020, 2021-06-19:
Remove power to the camera, insert the microSD card, turn the power back ON.

The yellow light will come ON and flash for roughly 30 seconds, which means the firmware is being flashed successfully. The camera will boot up.

-> my Camera do not start flashing...just rebooting

Johanson2020, 2021-06-19:
I tried that too, unfortunately without success.
https://github.com/TheCrypt0/yi-hack-v4/issues/327#
Firmware: 9.0.26.07_202104231123

closed 2021-06-21 by Johanson2020

alienatedsec, 2021-06-22:
@Johanson2020 this Camera is based on a newer chipset. It will not be compatible with v4 or v5