alienatedsec/yi-hack-v5 #87: Yi Home 1080p page does not load

ChenchaoZhao, 2021-07-21:
I installed the firmware for Yi Home 1080p.
The home page does not load. There is only a “HOME” showing up.

M0r13n, 2021-07-30:
I noticed the same behaviour. With older versions of the hack, it was possible to access the web page with just the hostname or ip address. But in this case, you have to explicitly access the camera on port `8080`.

So if your camera has the ip `192.168.1.24`, you would need to type: `http://192.168.0.24:8080` into your browser. At least for me this worked fine.

chassett1, 2021-08-06:
Good afternoon, 

I find I'm having the same issue with the home page not loading, even with the 8080 port.  I'm able to still access the camera on the YI app, so not all is a loss.  this is on version 0.3.1

pbix, 2021-08-08:
I was able to get my Yi Home 1080p 18US to load after some difficulty.

In my case I was upgrading from Yi-Hack-v4 after being apparently cheated out of my money.  I copied only the rootfs_y20 & home_y20 files to my root of my SD card because I did not read enough.  The lights all seem to flash appropriately but no web page on 80 or 8080.  Later I learned about and copied the required yi-hack-v5 directory to the SD card root but still no love.

The trick was to delete everything off of the SD card, put the Yi-Hack-v4 files on the SD card,  Downgrade to Yi-Hack-v4 and then put ALL the required files for Yi-Hack-v5 on the SD card and upgrade.  Finally access via 8080 is required as port 80 is by default for ONVIF.  You can change these defaults after you are in.

It seems you cannot recover otherwise from an incorrect flash procedure.  Hope this trail of tears can help some others struggling with getting Yi-Hack-v5 to work on their Yi Home 1080p.

chassett1, 2021-08-11:
Thank you for the quick reply.  What folder do we need to copy over? I only see the Home and Rootfs files in the zip files. Sorry if

chassett1, 2021-08-11:
I don't know why I didn't see the Yi-Hack_v5 folder in the first download I had, but it was there when I started from scratch again today... loaded and working great.  thank you for the help.

chassett1, 2021-08-12:
I just wanted to note that I was also installing the crypto Yi-Hack-v4 originally, and didn't realize alienatedsec had their own version.

ChenchaoZhao, 2021-08-14:
I tried @pbix 's method. it still didn't work. i pinged the camera ip and it was connected. then i tried ssh to it but it refused. the webpage did not load. so i guess the flash didn't work at all.

Fookes74, 2021-11-22:
I'm having the same issue on a Yi Outdoor camera. Bizarrely, having previously flashed a Yi Home camera successfully (and been able to access it via its webpage) I can no longer access that either. Both have different names, both are connected, on the network and working in the Yi Home app. Neither are working in Apple Home app. Neither are accessible via IP address. Both had 2 files and folder copied across to the root folder of the SD card. At a complete loss what to do now. Suggestions? 
Beginning to wonder how stable this all is.

martinezpenya, 2021-12-05:
> I'm having the same issue on a Yi Outdoor camera. Bizarrely, having previously flashed a Yi Home camera successfully (and been able to access it via its webpage) I can no longer access that either. [...]

Same problem here moving YIhome 47US from yihackv4 to yihackv5... I've only copied the rootfs_y18 and home_y18 files. When reading your problem I've downgraded to yihackv4... and then copy all files (including yihackv5 folder) and then upgrade was working well.
Thank you!

pmirshad, 2022-10-17:
I have a Yi Home 1080p (**18US**) that was running `2.1.0.0E_201809191630`. Installed _yi-hack-v5 - v0.3.4 - BETA_ today by following the steps listed on the main page. It worked flawlessly. Thank you so much :+1: 

```
/home/yi-hack-v5 # cat /proc/cpuinfo 
Processor	: ARM926EJ-S rev 5 (v5l)
BogoMIPS	: 269.10
Features	: swp half fastmult edsp java 
CPU implementer	: 0x41
CPU architecture: 5TEJ
CPU variant	: 0x0
CPU part	: 0x926
CPU revision	: 5

Hardware	: hi3518ev200
Revision	: 0000
Serial		: 0000000000000000

/home/yi-hack-v5 # free -m
              total        used        free      shared  buff/cache   available
Mem:             30          15           9           0           5           0
-/+ buffers/cache:           15          14
Swap:             0           0           0
```

closed 2023-07-15 by github-actions[bot]