alienatedsec/yi-hack-v5 #88: RTSP URL Insecure even with Password Set in Config Tab

aX1Mus, 2021-07-27:
Is there a way to force password authentication on the RTSP URLs?
The Authentication section of the Config tab mentions the username and password for RTSP, ONVIF, and HTTP, however, any connections to RTSP URL without credentials is successful.   If I can somehow lock down the RTSP via an IP or user I would like to move my ZoneMinder to GCP or AWS.  Is there something undocumented I am missing?

LinoBarreca, 2021-08-23:
Same here. Looks like a bug.

alienatedsec, 2021-10-05:
@aX1Mus password is not there yet but I am working on it

hudak0va, 2022-04-05:
same for me, any updates? @alienatedsec

alienatedsec, 2022-04-05:
@hudak0va I know that I have not been active for a while, but my MSc is now finished, and I will slowly get back to this project. I cannot promise anything this month as I am going on a long holiday (well deserved).

I also need to move my development rig and adopt updates and changes from the past six months before getting fully back into it - looking forward, but please be patient for a little while

alienatedsec, 2022-12-05:
The next release, `0.3.7(?)` is to support username and password on RTSP.

alienatedsec, 2023-07-07:
Resolved with https://github.com/alienatedsec/yi-hack-v5/discussions/248

closed 2023-07-07 by alienatedsec