alienatedsec/yi-hack-v5 #92: [YI Outdoor] Freezing

thommycgn, 2021-08-19:
Webinterface reading/saving takes long time. 
RTSP Stream is freezing
Snapshots don't work, camera hanging up
Hole Camera is not reacting any more, only power off/on is working

dojf, 2021-08-22:
Same sort of problems with the Yi outdoor cam (model 9CUS) :
Slow webinterface, no RSTP stream possible, I can get snapshots but it takes much time (~40 seconds) to appear. 
I installed a swapfile but no effect. Everything is very slow compared to my Yi Indoor Cams with the Mstar software.
Also switch to "Disable cloud" does not seem to work as my Yi App still finds the cam.

reef-actor, 2021-10-03:
Issue #47 might explain the freezing. I think a fix was created but no new releases have been made since

alienatedsec, 2021-10-05:
#63 is also related to swapping issue.

dojf, 2021-11-02:
@alienatedsec  Thanks, it took some time to test this, sorry, but it now looks like it's working fine. 
So thanks again !