alienatedsec/yi-hack-v5 #95: Yi-Dome 43US lost connection and now get wifi password error when trying to reconnect via the app

ronaldlw, 2021-08-29:
This camera was working but it dropped from zoneminder and has the blue light flashing.  Tried reset and reconnecting via the app but after putting in the wifi network and password and the camera successfully reading the QR code it finally fails with "wifi password error".  I'm sure the password is correct, I've tried a couple of routers with same results.  Suggestions?

ronaldlw, 2021-08-29:
I did have it set to disable the cloud, i'm not sure if that affects the re-registration process or not, if so is there a way to re-enable that by editing a setting on the sd card or something?  I also reformatted and copied the files back to the card again to see if that would help.

shelken, 2021-09-30:
same issue.

alienatedsec, 2021-10-05:
It would be best if you rolled back to factory firmware - then try #63 after applying the hack

I will release another version soon

closed 2023-07-14 by github-actions[bot]