alienatedsec/yi-hack-v5 #105: Bricked Yi 1080p Home 48US?

caffeineflo, 2021-10-05:
Hi!

The readme mentions that there is a way to unbrick cameras but says "ToDo" and nothing else yet. So I was hoping to get some more information about the process here.

My issue is that I have been using a brand new Yi 1080p Home 48US with this hack for about a month before the camera (without interaction) just started to be stuck in a reboot loop.
The camera comes on, orange light, then blue, then restart (hear the shutter open and close). This goes on forever.

I've reset the camera already, tried to reconnect "waiting to connect", but every time I show the QR code and the camera reboots it's back to "waiting to connect" now.

Hopefully someone can help

alienatedsec, 2021-10-05:
@caffeineflo 
Try [those files](https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/32280606850) (found via web search engines)

BigHomie90, 2021-11-12:
> Hi!
>
> The readme mentions that there is a way to unbrick cameras but says "ToDo" and nothing else yet. So I was hoping to get some more information about the process here.
>
> My issue is that I have [...]

I have exactly the same problem as you described. After a month with yi hack v5 endless boot loop started.. Have you find a solution to unbrick?

BigHomie90, 2021-11-12:
Followed @iMrLopez  and placed yi hack v4 files to sd and camera unbricked. Thanks!

caffeineflo, 2021-11-13:
> Followed @iMrLopez and placed yi hack v4 files to sd and camera unbricked. Thanks!

Exactly this did it for me! Thanks!

closed 2021-11-13 by caffeineflo