alienatedsec/yi-hack-v5 #108: What Yi Home apps version used to connect?

chilskater, 2021-10-19:
Hi,
I saw these hack while browsing internet recently. I have old Yi Ant 1080P camera with 18CN serial number. Few years ago manage to get it ran outside China using Chinese apps and modified firmware. It used proxy inside China then it didnt worked and kept under the drawer.
Now i managed to insall yi hack v4 or v5 but cant get it connected to any apps. I can ping, telnet etc...it said connected to wifi but didnt paired with the apps

hkrob, 2021-10-20:
The only real gap imho with not having the app is the inability to program pre-sets...
I have my cams feeding into frigate which is working well.

chilskater, 2021-10-20:
strangely i can access to the camera settings page but it didnt load anything except the menu only..it was okay yesterday

closed 2023-07-10 by github-actions[bot]