alienatedsec/yi-hack-v5 #116: Audio output

Ansarowsky, 2021-11-11:
Is it possible for camera to play any audio with built-in speakers?

alienatedsec, 2021-11-18:
It might not be something to develop instantly, but it is good to keep in mind

alienatedsec, 2024-02-14:
Please try `0.4.1i` @samariini @ansarowsky

alienatedsec, 2024-02-19:
This is added in the latest pre-release - please raise another issue if needed

Piceeeeeee, 2024-07-30:
cant seem to find the option to use the speaker, any help?

korayal, 2024-12-28:
@alienatedsec FYI, It appears  there's been a misunderstanding here, the ask here is the same as the one in https://github.com/alienatedsec/yi-hack-v5/issues/408

As opposed to `yi-hack-MStar`, there's no [`/tmp/audio_in_fifo`](https://github.com/roleoroleo/yi-hack-MStar/blob/9cf96db961f320970b8267acae4eabce592f9836/src/www/httpd/cgi-bin/speaker.sh#L126) in this fork (yet?)

Looks like that feature was [introduced into the MStar version of RTSPServer](https://github.com/roleoroleo/yi-hack-MStar/commit/e19bf1b349d7b132be520f31d93b9549ddf299be) 3 months after the version we have here.