alienatedsec/yi-hack-v5 #117: Possible to upgrade from Yi Hack V4 to this version?

Fookes74, 2021-11-12:
Hi

Is it possible to upgrade from Yi cameras running Yi Hack V4 to Yi Hack V5? If it's possible but risky I'd prob leave 'as is' but if it is possible without problem I'd like to do so.

Thanks

BigHomie90, 2021-11-12:
Format sd card and place files as instructed. Follow the next steps from the V5 guide to complete.

I tried to upgrade from v4 using user interface but no success so do it if you like by placing the files and folder directly to the root of your sd card.

PilaScat, 2021-11-15:
Remember that the web port for v5 is 8080, I tried more and more times to install it for this

ViktorSchlaffer, 2021-11-16:
> Hi
> 
> Is it possible to upgrade from Yi cameras running Yi Hack V4 to Yi Hack V5? If it's possible but risky I'd prob leave 'as is' but if it is possible without problem I'd like to do so.
> 
> Thanks

Yes, I just did it.

alienatedsec, 2021-11-18:
Please follow the guidance and make sure you copy the folder.

closed 2021-11-18 by alienatedsec