alienatedsec/yi-hack-v5 #120: Yi Dome 1080P and Yi Outdoor Cam - Ability to use 32Gb or 64Gb SD card?

Fookes74, 2021-11-20:
Hi,

I'm trying to flash Yi Hack V5 on to the above cameras, starting with Dome Cam with a 32Gb card in. When plugging in SD card and starting camera it just boots as normal. Tried repeated times without it taking. Is it the size of the SD card that is affecting it, and will it likely be the same for the Outdoor cam with 64Gb card in? 

The wording states 'preferable' 16Gb or less SD capacity but it doesn't insist on it. Can anyone shed light on this?

alienatedsec, 2021-11-21:
@Fookes74 the only rule is the card needs to be FAT32 to flash successfully

Fookes74, 2021-11-21:
> @Fookes74 the only rule is the card needs to be FAT32 to flash successfully

But that's exactly what I've done. Used a Mac via Disk Utility set to MS-DOS(FAT). Also used SD Formatter app. Tried on two different SD Cards (both 32Gb) on two different YI 1080p Dome cameras and both go solid yellow for around 5 seconds and then go to flashing blue. Never solid blue, and certainly not a decent period of yellow whilst it boots in to the hack. 

Any advice? Does the speed of card make a difference? Whilst I appreciate your comment I'm curious why the wording mention the preference for a 16Gb card or less?

alienatedsec, 2021-11-22:
@Fookes74 because those are usually formatted FAT32 by default in Windows

alienatedsec, 2021-11-22:
@Fookes74 i am not a MAC user so I cannot help.

Fookes74, 2021-11-22:
Anyone else have experience on this (via Mac)? I've flashed a Yi Home camera with a 16Gb card first time without issue. I'm beginning to wonder whether formatting SD cards of this capacity on Mac is causing the issues.

Z1mDMan, 2021-11-26:
> Anyone else have experience on this (via Mac)? I've flashed a Yi Home camera with a 16Gb card first time without issue. I'm beginning to wonder whether formatting SD cards of this capacity on Mac is causing the issues.

use disk utility, select the card, Format: MS-DOS (FAT) and Scheme: Master Boot Record
[image]

I have done this successfully with a 32Gig card. I haven't tried bigger yet.

ads1230, 2021-12-28:
Use the official app for formatting.

https://www.sdcard.org/downloads/formatter/sd-memory-card-formatter-for-mac-download/

jofemodo, 2022-12-11:
I'm trying on Yi Outdoor 1080p with 32GB Kingston (brand new).

+ First, with the preformatted partition => No luck
+ Deleting partition and recreating a primary partition of 32GB (type "b" on fdisk). Use mkfs.vfat to format. => No luck
+ Deleting partition and recreating a primary partition of 16GB (type "b" on fdisk). Use mkfs.vfat to format. => No luck
+ Same process using gparted => No luck

So, it seems that Outdoor doesn't work with 32GB SD-card. At less not with the Kingston i have ;-)
Tomorrow i will try with a smaller one.

Regards,

jofemodo, 2022-12-11:
Details from my camera:

+ **Product Name:** Yi Outdoor Camera 1080p
+ **Production Model:** YHS-3017
+ **DID:** RFUSY597BO9IQL220408
+ **S.N.:** RH30GAEUW25006696

Thanks!

alienatedsec, 2022-12-11:
@jofemodo too much detail in your post - don't expose yourself.

The first four digits ``xxUS`` and the current firmware version will be enough.

jofemodo, 2022-12-13:
OK! Thanks for your advice!

jofemodo, 2022-12-13:
I just tried to burn the firmware again, using a smaller SD-card, formatted as fat32 and copying the latest firmware files in the root, but it didn't work for me. The camera boots normally. No blinking orange. Am I missing dome stupid detail, or could my camera  not compatible with this? 

My camera is RFUS with a very recent firmware, from mid 2022.

Thanks

jofemodo, 2022-12-14:
Excuse @alienatedsec !

You undervote my comment but didn't explain what is wrong with it. 
I supose i didn't provide the right info. Let's try again ...

+ RFUSY597 => The first four digits "after" the xxUS
+ Firmware: 202205??

The firmware version is not complete, sorry. I missed to take with me the camera packaging and now it's 150km far from here, so i can't access this info until i come back. The label on the camera doesn't seem to contain this info. 

Thanks

alienatedsec, 2022-12-14:
It is the Allwinner platform, and is not compatible with v5; hence why. Try @roleoroleo repos

stevegoossens, 2023-02-06:
The section in this repo's readme (expandable with (Click) "spoiler") was written by me for v4 and is a method to make a small partition on a large capacity removable drive/card and works for the purpose of flashing this firmware.

closed 2023-07-07 by github-actions[bot]