alienatedsec/yi-hack-v5 #123: Cannot decide which firmware

vgezer, 2021-11-25:
I am unable to find the correct firmware for the camera

This is the info written on the label at the back of the camera:

Model: XF-XY43
Resolution: 1080P
SKU: Q3171US

I am using Yi Home App

Can I try e.g. Yi Home 17CN? Does it directly get bricked with wrong firmware?

Also a screenshot from Wireshark. It seems that the camera is using UDP port 16837

[image]

Any ideas?
Thanks.

phuonglm, 2021-11-27:
As i checked XF-XY43 using AK3918 SOC instead of hi3518ev200 like this repository targeted. So I'll said no support yet for  XF-XY43

vgezer, 2021-11-30:
Do you know if I can compile it myself using any of the available repositories? How hard would it be?

alienatedsec, 2021-11-30:
@vgezer it took me a few months, but I reckon it's possible if you have some spare time on your hands.

derpeter, 2022-06-19:
Hello,
I found this Kernel sources which seem to be for the ak3918 https://github.com/Lamobo/lamobo-d1 
It belongs to this kickstarter campain which means different camera but same SOC

closed 2022-09-28 by alienatedsec

Pierochr, 2023-01-09:
Do you find a compatible firmware?Q3171eu

AUTNeonine, 2023-08-03:
https://share.temu.com/u0sasTy4kfA

I am totaly confused now with this Yi Cameras now
It says Model: YIIOT-T27
Firmware: 6.0.05.10_202301061607

Im looking to integrate this to my HomeAssistant

Jolocode, 2023-09-20:
> > https://share.temu.com/u0sasTy4kfA
> > I am totaly confused now with this Yi Cameras now It says Model: YIIOT-T27 Firmware: 6.0.05.10_202301061607
> > Im looking to integrate this to my HomeAssistant
> 
> Any luck? I'm on the same mission you are. Trying to avoid a tear down or any destructive RE

**Push** - I'm also looking for an integration of this camera to my HomeAssistant. Did anyone find a solution already? Thanks for any helpful hints!

Schnup89, 2023-09-21:
I have the same camera, unfortunately it isn working (stuck after successfully connection to wifi).
I discovered a way to get root access via telnet:
For my camera the FTP-Service was enabled, so I added "telnet &" to the file /etc/jffs2/time_zone.sh
It was one of the few files that will be executed on startup and can be written via ftp.

Will check if I can unlock some features...

Schnup89, 2023-09-21:
https://ricardojlrufino.wordpress.com/2022/02/14/hack-ipcam-anyka-teardown-and-root-access/

It's seems that it is the same system

juliamarie66, 2024-04-15:
Which is the password for the ftp access?
Did anyone get this device hack?

juliamarie66, 2024-04-15:
Ok I got the password.

FredFeuerstein0815, 2024-06-13:
For FTP Username is root with no password.

Florin-Buzau, 2025-02-16:
> Ok I got the password.

i am in same situation, ftp is enabled but is no working with root and no password. Did you find the password?

juliamarie66, 2025-02-16:
Yes its working. I will check tomorrow what was the password

Florin-Buzau, 2025-02-17:
> Yes its working. I will check tomorrow what was the password
> 
> 16.02.2025 19:37:10 Florin-Buzau ***@***.***>:
> […](#)

I inserted even an sd card and i tried again with root without password but no luck. I am waiting for your reply, Many thanks in advance.