alienatedsec/yi-hack-v5 #124: Unable to pair to yi home app

weising26, 2021-11-25:
Is there any way to bypass the pairing like in v4? 
i have a cn version and able to connect to the web ui but unable to pair to the yi app even with proxychain enabled. after restart the cam will need to do the pairing process again to get the camera connected to wifi.

https://github.com/TheCrypt0/yi-hack-v4/issues/52

hkrob, 2021-12-16:
I have the same issue.
it appears proxychains doesn't work with yihack5

there does not appear to be a fix at this point.

rcplay, 2021-12-29:
I think that I have the same problem

closed 2023-07-05 by github-actions[bot]

MichlW, 2023-11-20:
I also think, that I have the exact same problem. Even with working proxychains-ng configuration I am unable to pair the camera with the chinese Yi Home app. I also think that proxychains is working correctly, because I do not hear the "...works only in china" voice.

I can not find a way to pair the cam with the app and I am still searching for the exact reason.

This is what I tried for now:
- Getting tha cam and the phone as close to the router as possible.
- resetting the camery by button.
- Deactivating any unusual service as suggested in [yi-hack-MStar/issues/52](https://github.com/roleoroleo/yi-hack-MStar/issues/52#issuecomment-591847174)
- Using different (but working!) proxies from http://free-proxy.cz/en/proxylist/country/CN/socks/ping/all