alienatedsec/yi-hack-v5 #125: 47US getting boot loop after sometime.

phuonglm, 2021-11-27:
After 1 or 2 days of use I reboot/ cut power/ WIFI AP restart then 47US camera got boot loop ( blue yelow bink - long yellow - led off - then reboot again. Inspect log.txt in `log_no_wifi_dev.tar.gz` I got this.
```
system call back libsns_ov9732.so_____________________
[./rmm][1/1/0:0:25:858]: sns_call_backsns_call_back
[./rmm][1/1/0:0:25:860]: HI_MPI_ISP_3A SUCCESS
[./rmm][1/1/0:0:26:131]: 
_____Gamma test for table 4__________
[./rmm][1/1/0:0:26:150]: 
---------Rock_COMM_VENC_BindVpss VeChn = 0, VpssGrp = 0, VpssChn = 0
[./rmm][1/1/0:0:26:156]: rc set success! venc(0)
[./rmm][1/1/0:0:26:159]: 
---------Rock_COMM_VENC_BindVpss VeChn = 1, VpssGrp = 0, VpssChn = 1
[./rmm][1/1/0:0:26:165]: rc set success! venc(1)
[./rmm][1/1/0:0:26:167]: rmm_init success[./rmm][1/1/0:0:26:169]: font:/home/base/iso8859-1-16x32.bin
[./rmm][1/1/0:0:26:170]: filesize=16384
[./rmm][1/1/0:0:26:183]: init font /home/base/iso8859-1-16x32.bin ptr size(4)[./rmm][1/1/0:0:26:184]: font:/home/base/iso8859-1-8x16.bin
[./rmm][1/1/0:0:26:186]: filesize=4096
[./rmm][1/1/0:0:26:188]: init font /home/base/iso8859-1-8x16.bin ptr size(4)[./rmm][1/1/0:0:26:190]: load_font success[./rmm][1/1/0:0:26:191]: msg_proc success[./rmm][1/1/0:0:26:192]: zbar_proc success[./rmm][1/1/0:0:26:196]: pid[928][./rmm][1/1/0:0:26:206]: pid[970][./rmm][1/1/0:0:26:207]: pid[973][./rmm][1/1/0:0:26:208]: pid[972][./rmm][1/1/0:0:26:209]: pid[974][./rmm][1/1/0:0:26:210]: pid[971][./rmm][1/1/0:0:26:213]: pid[975][./rmm][1/1/0:0:26:216]: rmm_vpp_create_bmp 1 success![./rmm][1/1/0:0:26:218]: rmm_vpp_create_bmp 0 success![./rmm][1/1/0:0:26:245]: AiGetLPCMStream start
 [./rmm][1/1/0:0:26:245]: pid[976][./rmm][1/1/0:0:26:265]: VencGetH264Stream vechn=1
 [./rmm][1/1/0:0:26:265]: VencGetH264Stream vechn=0
 [./rmm][1/1/0:0:26:464]: set ai volume via himm 0x201200d8 0x1e1ec001[./rmm][1/1/0:0:26:530]: set ai volume via himm 0x201200c8 0x23c2e[./rmm][1/1/0:0:26:633]: set ao volume via himm 0x201200d4 0x03122424[./dispatch][11/26/19:14:17:98]: DISPATCH_SET_DEFAULT_TIME 1637954057[./dispatch][11/26/19:14:23:700]: wpa may crashed, now reset
[./dispatch][11/26/19:14:23:850]: no usb dev
```

Since no`dmesg` log or serial connection I can find the root cause yet. But I guess maybe at that moment swap file is corrupted without working swapfile the next startup won't work.

Now I'm testing with this to make sure corrupted swap still fine at boot . Will report back in next week end. 
```
if [[ $(get_config SWAP_FILE) == "yes" ]] || [[ $MODEL_SUFFIX == "yi_dome" ]] || [[ $MODEL_SUFFIX == "yi_home" ]] ; then
    SD_PRESENT=$(mount | grep mmc | grep -c ^)
    if [[ $SD_PRESENT -eq 1 ]]; then
        if [[ -f /tmp/sd/swapfile ]]; then
            chmod 600 /tmp/sd/swapfile
            mkswap /tmp/sd/swapfile
            swapon /tmp/sd/swapfile
        else
            dd if=/dev/zero of=/tmp/sd/swapfile bs=1M count=64
            chmod 0600 /tmp/sd/swapfile
            mkswap /tmp/sd/swapfile
            swapon /tmp/sd/swapfile
        fi
        sysctl -w vm.dirty_background_ratio=2
        sysctl -w vm.dirty_ratio=5
        sysctl -w vm.dirty_writeback_centisecs=100
        sysctl -w vm.dirty_expire_centisecs=500
        sysctl -w vm.vfs_cache_pressure=200
        sysctl -w vm.swappiness=30
        echo 5 > /proc/sys/vm/laptop_mode
        echo 30 > /proc/sys/vm/swappiness
    fi
fi
```

phuonglm, 2021-12-04:
It did work. After 1 or 2 days when some thing wrong happened ( power cut, reboot, Access point disconnect ) the camera reboot it self and not able to start again. Boot then restart after lens click.
I tried remove sd card/ reset camera but still in boot loop. I guess something wrong saved to flash.

evilchen0815, 2021-12-11:
I have the same problem. As sometimes the RTSP hangs up, i rebootet the camera. After that it didnt came back and went into the boot loop. Only a new flash (older firmware, newer firmware) brings it back to life. its 5 time, i got this with different cameras (but all 47us)

phuonglm, 2021-12-12:
Got the serial connection to my 47US and here is the boot log when failed https://gist.github.com/phuonglm/6dc31f834156e1b1827419d4755120b3
```
ln: /etc/resolv.conf: No space left on device
```
I assume that some process write data to without clean up then make rootfs runout of space. I Tested this by created big file on rootfs then reboot then same problem happen.

Since free space left on rootfs pretty small so I will have to.
+ Find which process leave trash on rootfs and fix it.
+ If process is critical for me then I have to delete everything I don't need for on rootfs ( ie, ssh/ mqtt )....

phuonglm, 2021-12-13:
@evilchen0815 I found my problem and fix that but your problem maybe different with me. I have a script that continuous login (via telnet) to camera every 5 mins to sync camera status with my smart home. `/var/run/utmp` is the file which save log about login status and getting bigger after every login. And when it big enough wpa stop work because `no space left`. 

```
995.5K  bin
4.0K    dev
49.0K   etc
12.7M   home
674.0K  lib
512     linuxrc
512     mnt
0       proc
0       root
11.0K   sbin
0       sys
35.0K   usr
10.5K   var
```
Here is my `du -sh /*` after clean flash. You could try with your camera and see any different ( some directory may getting bigger than mind ) and find trash file in these.

phuonglm, 2021-12-18:
@alienatedsec could we have `/home/yi-hack-v5/.ash_history` and `/var/run/utmp` symlink to `/dev/null` in next release?

sordfish, 2021-12-19:
I've had node_exporter running on my camera for about a week. / is now stable with those two symlinks. Thanks!

mfizz1, 2022-01-29:
I too have the same issue. After about maybe a week or 2 of continuous use, the camera stops working continuous boot loop with or without sd card. I will sym link these 2 files and feedback.

chrizza87, 2022-04-06:
same problem on my side. added the two symlinks, but the used space is getting bigger. after a couple of weeks "/dev/root" expanded from 93% to now 96%. There seems to be another process, which "pollutes" the storage. 

Any suggestions or a hint where a can take a look? Perhaps even a cronjob for deleting files? Some investigative hint for me, would be awesome :-) 

Thx
chrizza

phuonglm, 2022-04-07:
> same problem on my side. added the two symlinks, but the used space is getting bigger. after a couple of weeks "/dev/root" expanded from 93% to now 96%. There seems to be another process, which [...]

Mine after add symlink, free space still reduce after sometime but no more boot loop. When free space reach 99% and camera reboot it will reclaim free space.  It seem related to garbage collector of JFFS2 https://en.wikipedia.org/wiki/JFFS2 .

chrizza87, 2022-04-14:
Thx for the hint. So if i would set up a cronjob to run once a month to reboot the camera, this will be the "solution"?

phuonglm, 2022-04-15:
I reboot it weekly via cronjob

alienatedsec, 2022-12-05:
Please upgrade to `0.3.6` to fix many other and similar issues

closed 2022-12-05 by alienatedsec